adafruit/circuitpython #2603: STM32: Crashing to HardFault_Handler after the second to CIRCUITPY after boot

siddacious, 2020-02-07:
This is a strange one: I've been iterating on some code on the Feather STM32F405 and today I'm reliably getting a crash into the hard fault handler when I do the following:
* power on/reset
* wait for `code.py` to start running
* write to CIRCUITPY -> triggers automatic restart
* wait for `code.py` to start running again
* write to CIRCUITPY
->Crash

Some context: I'm working on a project using the LPS33 and the Feather STM32F405 to make a sip & puff input device. I learned late yesterday that HID isn't supported on STM32 when I got and error complaining about USB being busy as soon as I added some HID code.

Prior to this error I had been writing to the CIRCUITPY drive regularly with no issue

I switched development to a metro M4 to continue working on the HID code, but when I started with the STM32 again this morning I started getting the hard faults/crashes.

My a zip of my current CIRCUITPY contents is attached

[sipnpuff.zip](https://github.com/adafruit/circuitpython/files/4172472/sipnpuff.zip)

siddacious, 2020-02-10:
@hierophect please let me know if/when you'll be able to look at this and if you have anything you'd like me to test. 

In the meanwhile I'll get CP onto another board to test and see if the behavior is replicated

hierophect, 2020-02-10:
I would like to get to the bottom of this - I think I've observed this issue sporadically in the wild, but I never had decent reproducibility. I will try your steps and see if I can get it to do the same.

tannewt, 2020-07-14:
@hierophect Are you able to reproduce this?

hierophect, 2020-07-14:
Kind of forgot about it, sorry. I will test this today, it may have gotten solved by some other PR or bugfix to flash, etc. If not we can discuss when I should investigate it.