adafruit/circuitpython #2604: Fomu port of CircuitPython

aidancuckoo, 2020-02-08:
The Tomu team have a port of CircuitPython [https://github.com/xobs/circuitpython/tree/fomu/ports/litex](url) for the Fomu (new to the Adafruit shop) so while I try to hunt down a Tomu port, you could add the port to the official CircuitPython repository. If I find a Tomu port, I will edit this post.

tannewt, 2020-02-11:
@xobs want to open a PR for official fomu support?

@aidancuckoo I don't know of a Tomu port. It only has 8k ram so using it won't be pleasant.

aidancuckoo, 2020-02-14:
@tannewt Apparently there was a mistake with the data import and so mithro, the creator of Tomu fixed the issue, therefore now according to the website, Tomu is not compatible with CircuitPython or even MicroPython.

gojimmypi, 2020-03-02:
@aidancuckoo do you have any additional details, link to more info, etc? You mention the `Tomu` (the ARM version) but this issue description is `Fomu'  (the FPGA version)

If **F**omu, this must be just regarding the hacker version, and not the production version that Adafruit sells, correct?

I'm certainly curious as to what @xobs has to say, as well.

gojimmypi, 2020-04-11:
fwiw - I have @tannewt 's Circuit Python working on my FOMU! :) I have a blog about it [here](https://gojimmypi.blogspot.com/2020/04/circuit-python-on-fomu-in-wsl-plus.html). In short, I used the [fomu toolchain](https://github.com/im-tomu/fomu-toolchain/releases) on WSL with the [LiteX port of Adafruit's Circuit Python](https://github.com/adafruit/circuitpython/tree/master/ports/litex).

dhalbert, 2021-06-15:
We now have a partial port. Closing this. More specific issues can be opened for additional functionality.

closed 2021-06-15 by dhalbert