adafruit/circuitpython #2609: Read/Write lots of data quickly

NAPtime2, 2020-02-10:
I am working with a Adafruit M0 Express, and have hooked it up to the Adafruit 16 bit I2C ADC ADS1115. I would like to read data at 250 Hz (permitted by the I2C ADS module) and save this data onto the flash on the M0 express. 

My trouble is if I just have a while True loop where I read data and write data in sequence, I am only able to record data a couple times a second (far short of 250 Hz). I suspect that this is due to the time it takes to write data, but don't know an awful lot about time limitations.

I can come up with partial workaround where I add the data to an list I keep in memory, and then save the list, but this leads to large gaps in the data during the time it takes to save the list. 

So my question is:
1. Is there a faster/more efficient way to save data quickly?
2. Is there a way I can be adding data to my array and saving it to flash at the same time?

tannewt, 2020-02-10:
Saving to flash can take time to erase and we currently block to do it. I'd suggest buffering to 512 byte chunks which will align to FATFS boundaries, it may make it a little faster. I don't think this will help with erase time unfortunately.

NAPtime2, 2020-02-10:
Thanks for the quick response. Do you know if saving to an SD card would speed things up?

tannewt, 2020-02-11:
I'd expect an SD card to be slower since the only interface we support is SPI. However, it could surprise me if the erase is faster.

NAPtime2, 2020-02-11:
I also have the STM32 feather, which I think uses SDIO for the SD card. Do you know if that is likely to be faster when that functionality is added to circuitpython?

tannewt, 2020-02-12:
It may be faster when SDIO is added. I don't really know and we have no immediate plans to add SDIO support.

tannewt, 2020-07-14:
Closing this in favor of other SDIO issue #3095

closed 2020-07-14 by tannewt