adafruit/circuitpython #2610: Issues with I2C in the REPL and scripts

siddacious, 2020-02-11:
I'm having issues setting up I2C in the REPL:
[image]

As you can see, in the REPL both methods of setting up I2C fail with complaints about the bus being busy. In the same screenshot you can see the sketch I'm testing working fine using `board.I2C()`

I discovered this in trying to refactor that code to use `busio.I2C(board.SCL, board.SDA)`; it complains about SCL being busy.

My code is here: mainly `code.py` and `puff_detector.py`

dhalbert, 2020-02-11:
I think you meant to post the files?

dhalbert, 2020-02-13:
@siddacious Could you post the files? Thanks.

siddacious, 2020-02-14:
@dhalbert sorry, here you go:
https://github.com/adafruit/Adafruit_Learning_System_Guides/tree/master/CircuitPython_Sip_and_Puff

dhalbert, 2020-02-19:
After a quick look, I'm thinking this is due to the I2C pins not being released on soft reload because the display is in use on I2C. The I2C singleton may be getting freed but displayio still has hold of the pins. The singleton should not be be freed and re-instantiated. Will need more investigation.

dhalbert, 2020-02-19:
@siddacious I'm trying to duplicate this, and not succeeding. I have a `code.py` that I think is the essence of your display setup. I have a 128x32 OLED, not the 128x64, but I don't that that should matter.
```python
import board
import displayio
import adafruit_displayio_ssd1306

displayio.release_displays()
i2c = board.I2C()
display_bus = displayio.I2CDisplay(i2c, device_address=0x3c)
display = adafruit_displayio_ssd1306.SSD1306(display_bus, width=128, height=32)

splash = displayio.Group()
display.show(splash)

while True:
    pass
```
I let this start up, then I ctrl-C it, and try this in the REPL:
```
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit Metro M4 Express with samd51j19
>>> import board
>>> i2c = board.I2C()
```
Does this match what you're trying to do? I don't have an I2C sensor attached, but I'm thinking that should not matter here. I do see the REPL on the OLED after I ctrl-C.

siddacious, 2020-02-20:
@dhalbert That wasn't quite what wasn't working for me but I think I've narrowed down how to reproduce it:
```python
import board
import busio

# Step 1: run the code as is, should output scan data

# Step 2: comment out the i2c init using board.I2C()
i2c = board.I2C()

# Step 3: Uncomment this i2c init using busio.I2C and save to reload
#i2c = busio.I2C(board.SCL, board.SDA)

i2c.try_lock()
for res in i2c.scan():
    print("Found address:", hex(res))
i2c.unlock()
```
as far as I can tell, `board.I2C` isn't letting the pin(s) go

siddacious, 2020-02-20:
@dhalbert I forgot to mention that if you pull power/reset between using `board.I2C` and `busio.I2C` it works fine

dhalbert, 2020-02-20:
In the example above, do you still have the `code.py` running? The display is going to grab and hold the I2C pins between soft reloads, because it is using the display for REPL output. So the general idea is to always use `board.I2C()` in your code instead of `busio.I2C(board.SCL, board.SDA)`. The latter will fail if a display needs the I2C bus so it can exist across soft reloads.

Testing without a `code.py`:

```
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit Metro M4 Express with samd51j19
>>> import board,busio
>>> i2c = board.I2C()
>>> i2c.try_lock()
True
>>> i2c.scan()
[60]
>>> i2c.unlock()
>>> 
soft reboot

Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit Metro M4 Express with samd51j19
>>> import board,busio
>>> i2c = busio.I2C(board.SCL, board.SDA)
>>> i2c.try_lock()
True
>>> i2c.scan()
[60]
>>> 
```

siddacious, 2020-02-20:
Gonna close this; If there _is_ a bug, it's hard to hit. If I can narrow down the case we can discuss re-opening. Currently I can only reproduce by involving my sip and puff code initially

closed 2020-02-20 by siddacious