adafruit/circuitpython #2611: cp: cannot create regular file Permission denied

CarlFK, 2020-02-11:
my script to setup a CPX with my custom config trips something that results in:
`cp: cannot create regular file '/media/carl/CIRCUITPY/Welcome_to_CircuitPython_CPX/.git/objects/12/27f7b26d3603eeb4e1dc44dd6a86918ac01801': Permission denied`

to reproduce,  
`git clone https://github.com/CarlFK/cpx`
wipe the cpx: >>> import storage; storage.erase_filesystem()
run this twice:
`./cpx_setup.sh`
`./cpx_setup.sh`

I am not reporting here to get help setting up my cpx, I'm reporting because it seems there is some bug in the cp fs gizmo.

dhalbert, 2020-02-11:
Script is here: https://github.com/CarlFK/cpx/blob/master/cpx_setup.sh

sommersoft, 2020-02-11:
> run this twice:
`./cpx_setup.sh`
`./cpx_setup.sh`

I can reproduce this by just running the `cp -a .../.git/objects/....` commands twice, as is done in the "offending" line: https://github.com/CarlFK/cpx/blob/6e019b231d948c3642b9b19f3d59276cf6394e02/cpx_setup.sh#L52.

I'm pretty sure this is related to using the `-a/--archive` option with `cp`. Here is the stat on a file in `.git/objects` (which I used to reproduce):
```
File: .git/objects/1e/631c4691c07f3b098804f2b2bab42456bd27a5
  Size: 3746            Blocks: 8          IO Block: 4096   regular file
Device: 802h/2050d      Inode: 14817700    Links: 1
Access: (0444/-r--r--r--)  Uid: ( 1000/sommersoft)   Gid: ( 1000/sommersoft)
Access: 2020-02-11 16:08:55.026141921 -0600
Modify: 2020-01-21 18:14:50.313895151 -0600
Change: 2020-01-21 18:14:50.313895151 -0600
 Birth: -
```
By copying the file, and preserving the file attributes, it will set those objects as read-only. Which is why the second attempt to copy/overwrite triggers a permissions error.

If I drop `-a`, I get no permissions errors with repeatedly copying the same `.git/objects` file.

CarlFK, 2020-02-13:
Opps, my bad.  This isn't related to CP at all.
I get the same behaviour on my laptop's fs.

closed 2020-02-13 by CarlFK