adafruit/circuitpython #2622: Documentation example for displayio does not work.

geekguy-wy, 2020-02-14:
I worked with displayio on my PyGamer a bit and discovered that the example on this [page](https://circuitpython.readthedocs.io/en/4.x/shared-bindings/displayio/OnDiskBitmap.html) does not work.

I worked on this example with my PyGamer and have a new version of the script that *does* work:[code]import board
import displayio
import time
import pulseio

board.DISPLAY.auto_brightness = False
board.DISPLAY.brightness = 0
splash = displayio.Group()
board.DISPLAY.show(splash)

with open("/sample.bmp", "rb") as f:
    odb = displayio.OnDiskBitmap(f)
    face = displayio.TileGrid(odb, pixel_shader=displayio.ColorConverter(), width=160, height=128, x=0, y=0)
    splash.append(face)
    board.DISPLAY.show(splash)

    # Fade up the backlight
    for i in range(100):
        board.DISPLAY.brightness = 0.01 * i
        time.sleep(0.05)

    # Wait forever
    while True:
        pass[/code]

makermelissa, 2020-02-14:
Tested on 4.1.2 (this is the branch of documentation you are referencing) and the code worked fine.

makermelissa, 2020-02-14:
Tested the "latest" branch on 5.0 beta 5 and _that_ is not working.

PR #2623 (2020-02-14): Updated OnDiskBitmap RTD example for 5.x
Fixes #2622.
closed 2020-02-14 by makermelissa