adafruit/circuitpython #2627: Same wave file sounds different played on different Mixer voices

FoamyGuy, 2020-02-14:
I first noticed this issue with a [larger project](https://github.com/FoamyGuy/NeoTrellis-M4-CircuitPython-Music-Fun) when I upgraded a NeoTrellis M4 from 4.1.2 to cp 5 beta-5. With that project I found that I hear a pop sound when the file starts on voice 0 but no other voices. I eventually shifted all voices up by 1 skipping `0` and hear no artifacts.

I've narrowed it down to this code which runs on the CLUE:
```
import board
from audiopwmio import PWMAudioOut as AudioOut
import audiomixer
import audiocore
import digitalio
import time

note = audiocore.WaveFile(open("c4.wav", "rb"))

audio = AudioOut(board.SPEAKER)
mixer = audiomixer.Mixer(voice_count=7, sample_rate=8000,
    channel_count=2, bits_per_sample=16, samples_signed=True)

audio.play(mixer)

print("playing")
mixer.play(note, voice=0, loop=True)
time.sleep(1.0)
print("stopping")
mixer.stop_voice(0)

time.sleep(1.0)
mixer.play(note, voice=1, loop=True)
time.sleep(1.0)
print("stopping")
mixer.stop_voice(1)

time.sleep(1.0)
mixer.play(note, voice=0, loop=True)
time.sleep(1.0)
print("stopping")
mixer.stop_voice(0)
```
The c4.wav [file used is here.](https://github.com/FoamyGuy/NeoTrellis-M4-CircuitPython-Music-Fun/raw/master/wave_generator/sine/C4.wav)

This code plays the same file 3 times in succession on voices `0` then `1` then `0`. When I run in on the CLUE I hear 3 one with "choppiness" at the beginning, one with "choppiness" at the end, and one clean playback, in this order consistently.

I've also tested slight variation of this code (using `AudioOut` instead of `PWMAudioOut`) on SAMD51 based devices and found similar issues. Though no two devices I've tested so far produce the exact same sound artifacts. I've tested on PyGamer and NeoTrellis M4.

ladyada, 2020-02-14:
Clue uses PWM not analog output - it will never sound as good or clean as the SAMD51 or SAMD21! :)

FoamyGuy, 2020-02-15:
Ah I see, thank you. I will close this issue and post one with code and outcome for the SAMD51 based devices.

closed 2020-02-15 by FoamyGuy

kevinjwalters, 2020-07-27:
@jepler has mentioned voice 0 is processed differently to rest.

I'm about to file a bug ticket for voice 0 being noticeably louder than the rest and was looking to see if that was already covered by an existing issue ticket.