adafruit/circuitpython #2628: a  bug not to worry about.

ThomasAtBBTF, 2020-02-15:
Hi,

while playing around with Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit ItsyBitsy M4 Express with samd51g19 I did the following:
 1. import microcontroller
 2. dir(microcontroller.cpu)
This results in a frozen repl !
And after a while, the Windows serial device "goes away"...

I don't consider this at all serious, but worth to notice.

Thomas

tannewt, 2020-02-18:
This sounds like a good issue for someone who wants to learn how to use a debugger. My guess is that CircuitPython is either crashed or in an infinite loop. Here is a Learn guide to start with: https://learn.adafruit.com/debugging-the-samd21-with-gdb The same process works on the SAMD51 too.

DavePutz, 2020-06-18:
Looks like this one may be fixed by asf4 PR #34 for issue #2893. After putting in that patch the dir(microcontroller.cpu) returns correctly.

tannewt, 2020-06-18:
Thanks @DavePutz! dir may be incorrectly trying to read the voltage value rather than returning the name. What happens when you dir a `digitalio.DigitalInOut` object? It has properties like pull and direction that may not work.

DavePutz, 2020-06-18:
digitalio seems to work OK without the patch:
```
>>> import digitalio
>>> dir(digitalio)
['__class__', '__name__', 'DigitalInOut', 'Direction', 'DriveMode', 'Pull']
>>> a = digitalio.DigitalInOut
>>> dir(a)
['__class__', '__enter__', '__exit__', '__name__', 'deinit', 'direction', 'drive_mode', 'pull', 'switch_to_input', 'switch_to_output', 'value']
```
Without the patch dir(microcontroller.cpu) just hangs, stuck in the while loop in adc_sync_read_channel()
With the patch it looks OK:
```
>>> import microcontroller
>>> dir(microcontroller.cpu)
['__class__', 'frequency', 'temperature', 'uid', 'voltage']
```

DavePutz, 2020-06-18:
@tannewt , to be more complete, the stack leading up to the call of adc_read_channel looks like:
```
 	firmware.elf! adc_sync_read_channel (struct adc_sync_descriptor * const descr, struct adc_sync_descriptor * const descr@entry, const uint8_t channel, const uint8_t channel@entry, uint8_t * const buffer, uint8_t * const buffer@entry) Line: 122	
 	firmware.elf! common_hal_mcu_processor_get_voltage Line: 312	
 	firmware.elf! mcu_processor_get_voltage (mp_obj_t self) Line: 115	
>	firmware.elf! mp_convert_member_lookup (mp_obj_t self, const mp_obj_type_t * type, mp_obj_t member, mp_obj_t * dest) Line: 1042	
 	firmware.elf! mp_load_method_protected (mp_obj_t obj, qstr attr, qstr attr@entry, mp_obj_t * dest, mp_obj_t * dest@entry, _Bool catch_all_exc, _Bool catch_all_exc@entry) Line: 1118	
 	firmware.elf! mp_builtin_dir (size_t n_args, const mp_obj_t * args) Line: 193	
...
```

tannewt, 2020-06-18:
Huh, interesting. I'd consider this fixed then. I think it's a bit of weirdness around properties but nothing to worry too much about.

Thanks @DavePutz !

closed 2020-06-18 by tannewt