adafruit/circuitpython #2630: Teensy 4.0 - Output PWM

lukasloetkolben, 2020-02-18:
Hi i want to use PWM-out with my teensy 4.0 to dimm some LEDs.
I can't get it to work, i get an "invaild Pin"-Error. Is PWM already available on Teensy 4?

ladyada, 2020-02-18:
hi - you would want to comment/subscribe to this issue for updates when PWM is available :)
https://github.com/adafruit/circuitpython/issues/2489

closed 2020-02-18 by lukasloetkolben

arturo182, 2020-02-18:
PWM is not fully supported yet, but it should fail later, which pin are you using?

lukasloetkolben, 2020-02-18:
> PWM is not fully supported yet, but it should fail later, which pin are you using?

I am trying to use Pin 14 & 15

arturo182, 2020-02-18:
Yes, those don't support PWM output, but 0-9 (and more) should, but as I said, it will just fail later in the code :sweat_smile:

lukasloetkolben, 2020-02-18:
But those Pins should support PWM. In C++ i have no Problems with it. 
Do you have any ideas when the PWM Output is possible. 

I am not very in to the circuitPython project. Can you explain the problems you have by now?

yimidi-kr, 2020-04-28:
Still PWM is not supported.  Library is supported when use pin 0-9. But, Output is not generated!!!
Please check this issue^^ I tested on version 5.2.0