adafruit/circuitpython #2632: displayio.Palette: directly support 'None' to indicate transparent pixels

jepler, 2020-02-18:
Right now, it's necessary for every class that wants to deal with a color-or-transparent to check for a "None" value and use it to set the palette entry's transparent flag.  If "None" was supported by Palette's getitem/setitem in the core C code, then all this using code could be simplified and transparent color would always be available anywhere a numeric color is accepted.

Caveat: this means that displayio now parses a color that neopixel doesn't -- but a transparent neopixel isn't really a thing.

dhalbert, 2020-02-19:
@jepler did you want this to be in 5.0.0?

jepler, 2020-02-19:
I wouldn't hold back a release for this, it's a "would be nice" in my opinion.