adafruit/circuitpython #2636: allow None args for "end" values in busio/bitbangio I2C and SPI, or change doc

dhalbert, 2020-02-19:
The inline doc for read and write operations in `{busio,bitbangio}.{I2C,SPI}` says that the `end` values may be `None`, in which case they default to `len(buf)`. However, the default value is really `INT_MAX`. If you give `None` explicitly, you'll get an error.

Related to https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/pull/45.

Either `None` should be handled or the doc should be hanged.

tekktrik, 2022-05-01:
@dhalbert I think you fixed this with this commit, right? Or is it outstanding? https://github.com/adafruit/circuitpython/commit/b48bb75e24ca8502a0950626ae80e8c681049da4

dhalbert, 2022-05-01:
Right! Thanks, fixed by #5439.

closed 2022-05-01 by dhalbert