adafruit/circuitpython #2637: Errror loading 5.0 Beta 5 onto NRF52840 Feather Express

ATMakersBill, 2020-02-19:
Got this error while copying the .UF2 file onto a NRF58240 in Bootloader mode.
CIRCUITPY drive showed up after this, but I thought I should report it.

Windows 10, got the '840 about 2 months ago.

[image]

dhalbert, 2020-02-19:
What is in INFO_UF2.txt? I want to duplicate with the same bootloader version.

ATMakersBill, 2020-02-19:
UF2 Bootloader 0.2.6 lib/nrfx (v1.1.0-1-g096e770) lib/tinyusb (legacy-525-ga1c59649) s140 6.1.1
Model: Adafruit Feather nRF52840 Express
Board-ID: NRF52-Bluefruit-v0
Bootloader: s140 6.1.1
Date: Dec 21 2018

dhalbert, 2020-02-19:
I can duplicate this with "UF2 Bootloader 0.2.6", but it works fine with 0.2.9. So just update the bootloader if you want to stop seeing this.

This seems similar to #1575, which was fixed in the bootloader by fixes pertaining to https://github.com/adafruit/Adafruit_nRF52_Bootloader/issues/46.

closed 2020-02-19 by dhalbert