adafruit/circuitpython #2639: Sanity check memory use in 5.0.0

tannewt, 2020-02-19:
re: https://forums.adafruit.com/viewtopic.php?f=60&t=162220

A hackspace magazine example no longer works due to 30k more memory use.

Would someone be willing to replicate this? Thanks!

ladyada, 2020-02-19:
@jepler wanna take a look?
im suspicious of the bitmap, maybe it wasnt converted to 8-bit indexed?

jepler, 2020-02-23:
@ladyada I can take a look on Tuesday if that's soon enough

ladyada, 2020-02-23:
yep!

dhalbert, 2020-02-24:
I started looking at this. I get the identical memory issue when the program runs automatically as `code.py`. If I import `code.py` in the REPL it runs fine.

`otter_data.py` has a hideously large set of nested lists of small integers and could be made a lot more efficient.

Here is an instrumented view of memory consumption:
```
import board
import gc
print("before most imports", gc.mem_free())
import displayio
print("a", gc.mem_free())
from time import sleep
import otter_data
print("b", gc.mem_free())
display = board.DISPLAY
group = displayio.Group()
print("c", gc.mem_free())
otter_bitmap = displayio.Bitmap(384,96,18)   # <---- fails here
print("d", gc.mem_free())
background_file = open("/images/background.bmp", "rb")
background_bitmap = displayio.OnDiskBitmap(background_file)
background_tile = displayio.TileGrid(background_bitmap, pixel_shader=displayio.ColorConverter())
#...
```
Automatic run:
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.
code.py output:
before most imports 220080
a 220064
b 32000
c 31904
Traceback (most recent call last):
  File "code.py", line 13, in <module>
MemoryError: memory allocation failed, allocating 36864 bytes
```

Manual import of code.py:
```
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit PyPortal with samd51j20
>>> import code
before most imports 219744
a 219728
b 70336
c 70240
d 33328
```

dhalbert, 2020-02-24:
I bisected and got to this commit as the first one that causes the problem: d32dc814d3f0d3e845e54a043467f37a5908f5f0, which is #2330, which fixes a GC bug in displayio, _but supposedly only for epaper_. I repeated the bisect and got the same thing. Moving back one commit, I don't see the issue. So this would seem to make no sense, because I'm testing with a pyportal, which is not epaper. But maybe there is a mixup somewhere in displayio about epaper vs not?

Tagging @tannewt in case this rings a bell. Taking a break now.

tannewt, 2020-02-25:
Nothing jumps out at me. My next step would be to see if I can break in that collect via a debugger. Let me know if you'd like me to take a look.

dhalbert, 2020-02-25:
Yes, I'll look at this tomorrow morning: set some breakpoints and see what's going on. It shouldn't be calling the epaper code at all.

dhalbert, 2020-02-25:
This was not working on beta.5, but is working on tip of master now. :upside_down_face: More later.

tannewt, 2020-02-25:
Are we using the inline limits at all on this board? I wonder if the compiler is improperly dropping the type check sometimes.

dhalbert, 2020-02-25:
No, pyportal has no tricky stuff in mpconfigboard.*

dhalbert, 2020-02-25:
I am doing some more bisects, but I'm suspicious of something uninitialized somewhere.

dhalbert, 2020-02-25:
I did a bisect from 5.0.0-beta.5 to the tip of master. `git bisect` assumes you are going from "good" to "bad", not the other way, so I had to mark good results as bad and vice versa. (I didn't know this at first and got into some confusing situations.) The problem got "fixed" at 36e6cc8febf7f744f61772d29eed68d9ebb57200 (#2614), which makes sense because it allocated memory more efficiently, but it didn't necessarily fix any underlying bug. Will do a little more looking on beta.5.

dhalbert, 2020-02-25:
5.0.0-beta.5 gets the memory error only if compiled with `-flto` and with inlining enabled. Disabling `-flto` or enabling `-fno-inline` causes the problem to go away. And the breakpoint addresses are very fishy in the problem case:
```
(gdb) bre displayio_display_collect_ptrs
Breakpoint 3 at 0x15280: file ../../shared-module/displayio/display_core.c, line 311.
(gdb) bre displayio_epaperdisplay_collect_ptrs
Breakpoint 4 at 0x15282: file ../../shared-module/displayio/display_core.c, line 311.
```

dhalbert, 2020-02-26:
I spent time looking at the assembly code for 5.0.0-beta.5, and for e560b419f, which is the last commit before beta.5 that does not show the problem. I don't see anything wrong with the code. It did a ton of inlining: in both, all of `displayio_gc_collect()` is inlined, and all the single-use functions it calls are inlined. In e560b419f, even more lower-level functions are inlined.

I think the difference here may have something to do with how gc handles pointers in existing stack frames. Perhaps in 5.0.0-beta.5 case, less of what is on the stack may be found to be garbage at that time.

Given that the latest master has better behavior, due to #2614, I think it's not worth pursuing this further, at least for now.

The original Hackspace example could really be cleaned up: the huge nested list of lists could be rewritten as a list of `bytes()` similar.

I'm fine with closing this for now. @tannewt do you concur?

tannewt, 2020-02-26:
@dhalbert I'm fine closing it. I've seen similar with pointers in stack frames living longer than intended. It's all part of the imprecision in the GC process.

closed 2020-02-26 by tannewt

kevinjwalters, 2020-06-18:
One other unusal aspect of this program is the line lengths in `otter_data.py`. The first line has 113326 characters on it which might make it a record breaker for CircuitPython.

```
$ awk '{ print NR, length($0) }' otter_data.py
1 113326
2 23
3 10
4 10
5 10
6 10
7 10
8 10
9 10
10 10
11 10
12 10
13 10
14 10
15 10
16 10
17 10
18 10
19 10
20 10
21 1
```

I'd be tempted to `mpy-cross`, verify the output and then see what happens if I wasn't stuck in my own `MemoryError` hell.