adafruit/circuitpython #2646: PR #2644 (disable remote editing) is breaking BLE

dhalbert, 2020-02-21:
I was trying a BLE program on the latest master, and ran into this:
```python
>>> from adafruit_ble.services.nordic import UARTService
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "adafruit_ble/services/nordic.py", line 38, in <module>
  File "adafruit_ble/services/nordic.py", line 50, in UARTService
  File "adafruit_ble/uuid/__init__.py", line 74, in __init__
_bleio.BluetoothError: Unknown soft device error: 0002
```
A build from the commit just before #2644 doesn't have this problem.

Tagging @tannewt; I'll investigate too.

dhalbert, 2020-02-21:
Error 0002 is SD not enabled. The failure is in UUID creation. So the SD was enabled previously before #2644, but is now not enabled implicitly.

tannewt, 2020-02-21:
Sorry about this! I only tested the advertising demo I'm working on. Will look at the PR now.

PR #2647 (2020-02-21): Enable _bleio adapter when _bleio is imported
Fixes #2646.

Uses `MICROPY_MODULE_BUILTIN_INIT` mechanism to call a native module `__init__()` function when the `_bleio` module is imported. The init function enables the `Adapter`, which enables the SD. This mechanism was disabled in all our ports before this PR, but might come in handy in other cases.

Also fixed one minor doc thing I found by accident.
closed 2020-02-21 by tannewt