adafruit/circuitpython #2650: from adafruit_ble.uart import UARTServer not compatible with CircuitPython v.x

itpvga, 2020-02-23:
running the libs from the latest bundle to use with CircuitPython 4.x.x:
adafruit-circuitpython-bundle-4.x.mpy-20200222.zip
adafruit-circuitpython-pca10059-en_US-4.1.2.uf2

Loading the first libraries:
import board
from pulseio import PWMOut
from adafruit_ble.uart import UARTServer
from adafruit_bluefruit_connect.packet import Packet

Result in the following error code:

code.py output:
Traceback (most recent call last):
  File "code.py", line 3, in <module>
  File "adafruit_ble/__init__.py", line 33, in <module>
ImportError: This release is not compatible with CircuitPython 4.x; use library release 1.x.x

dhalbert, 2020-02-23:
If you want to use 4.1.2, then use release 1.0.2 of the adafruit_ble library: https://github.com/adafruit/Adafruit_CircuitPython_BLE/releases/tag/1.0.2. But we hope to release 5.0.0 for general availability quite soon, with release-candidate releases starting this week. So do consider moving to 5.0.0-beta.5 right now.

Right now we don't have a way of customizing the bundles so that the 4.x and 5.x bundles are guaranteed compatible with their respective release. It's not an easy problem to solve, unfortunately.

itpvga, 2020-02-23:
Thanks, the error message is indeed gone now.

However running the Bluefruit Connect app using my Android, the following message appears when trying to connect:

code.py output:
Traceback (most recent call last):
  File "code.py", line 11, in <module>
  File "/lib/adafruit_ble/uart.py", line 76, in start_advertising
OSError: Failed to start advertising, err 0x0012

I guess the best way forward is going for the 5.0.0 beta release...

ladyada, 2020-02-23:
5.x is where a ton of BLE work got done, we recommend going with the latest build/beta release - it'll be release-candidate very soon and its stable :)

dhalbert, 2020-02-23:
The API has changed a lot, so you'll need new versions of the libraries, and also you'll need to update your own program.

itpvga, 2020-02-23:
I got it working with the beta version...thanks for the support. Looking forward to the official 5.0.0 release

closed 2020-02-23 by itpvga