adafruit/circuitpython #2660: MKR Zero and Nano 33 IoT released as UF2

fgallaire, 2020-02-28:
They needs to be released as BIN

fgallaire, 2020-02-28:
Shouldn't apply to Circuit Playground Express and Trinket M0 (and all M0 board) too as I don't understand how 488 KB and 359 KB UF2 can be flashed on 256 KB MCU ?

dhalbert, 2020-02-28:
The UF2 files are not densely packed. Each 512-byte block consists of at most 476 bytes of data, and possibly less: https://github.com/microsoft/uf2.

There are UF2 bootloaders available for `mkrzero` and `nano33iot` in https://github.com/adafruit/uf2-samdx1/tree/master/boards, but you are right, we should publish `.bin` as well.

fgallaire, 2020-02-28:
OK thanks @dhalbert for the explanation

fgallaire, 2020-02-28:
> There are UF2 bootloaders available for `mkrzero` and `nano33iot` in https://github.com/adafruit/uf2-samdx1/tree/master/boards, but you are right, we should publish `.bin` as well.

Yes, but I see no Arduino boards in the releases

> The UF2 files are not densely packed. Each 512-byte block consists of at most 476 bytes of data, and possibly less: https://github.com/microsoft/uf2.

So, if the .bin file used to generate the UF2 file is under 256 KB, the UF2 file is always goof to be flashed regardless its real size ?

tannewt, 2020-07-14:
They are now released as bin and uf2.

PR #2664 (2020-03-01): Download links from S3; Do not upload release assets to GitHub
We now have over 1700 release assets. GitHub releases don't handle this many assets well:
- The release pages are huge and take a long time to load.
- Uploading assets can fail sporadically or hit GitHub rate limits.
- It's difficult to correct the problems above: the release action must be re-run several times and may still continue to fail. Uploading the missing assets is tedious.

We already upload the release assets to S3 without difficulty. This PR changes the https://circuitpython.org/downloads download links to point to a Cl [... cut]
closed 2020-07-14 by tannewt