adafruit/circuitpython #2667: Setting led values on CLUE sometimes takes hundreds of milliseconds

kevinjwalters, 2020-02-29:
Trouble at mill:

```
Adafruit CircuitPython 5.0.0-beta.5 on 2020-02-05; Adafruit CLUE nRF52840 Express with nRF52840
>>> from adafruit_clue import clue
>>> import time
>>> print(time.monotonic())
38.242
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.386993
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>>
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.00100708
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.00100708
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.000976563
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.00100708
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.00100708
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.00100708
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.white_leds = True ; clue.white_leds = False ; print(time.monotonic()-t1)
[... 150 lines cut ...]
0.00100708
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.000976563
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0010376
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.000976563
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.276001
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0010376
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.273987
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.000976563
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0010376
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.000976563
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.273987

```

dhalbert, 2020-02-29:
5.0.0-rc.0 is different because of https://github.com/adafruit/circuitpython/pull/2601. Since you're running on the CLUE, this may be due to displayio and display refresh. Try the same thing on a board without a display.

kevinjwalters, 2020-02-29:
I had a quick play with `board.DISPLAY.auto_refresh=False` and I've not managed to get it to go slow with refresh off...

kevinjwalters, 2020-02-29:
Is this just because of the very slow scroll of the screen and that those screen updates can happen in between any python instruction even ones that aren't outputting to the screen?

ladyada, 2020-02-29:
you could try releaseing the display

kevinjwalters, 2020-02-29:
The manually initiated refresh does measure at the same value.

```
>>> board.DISPLAY.auto_refresh=False
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; clue.red_led = True ; clue.red_led = False ; print(time.monotonic()-t1)
0.0
>>> t1=time.monotonic() ; board.DISPLAY.refresh(minimum_frames_per_second=0) ; print(time.monotonic()-t1)
True
0.272461
```

kevinjwalters, 2020-02-29:
Could this also be a possible explanation for `clue.color()` data freezing when using it in REPL mode? In the unlucky case the _long_ pause for a display refresh might occur during some i2c operations in python code. I'm wondering if 200-400ms could be long enough to break stuff there?

kevinjwalters, 2020-03-02:
My comment from discord:

> Returning to [this issue]- is that "just the way things are" if one is using `displayio` ? I'm wondering if it's problematic to introduce these _long_ pauses between any python statements if that's how it's working? I put a comment in there about i2c code in python. I'd assume that the previous "background" tasks like GC introduced delays in the 1-10ms range? Hundreds of ms sounds trickier. Is there a way of slicing that up or letting the application programmer choose where it can happen?

kevinjwalters, 2020-03-02:
Faster refresh (272ms) with rc0 probably due to 4x speedup of SPI as mentioned on Discord.

kevinjwalters, 2020-03-02:
My Python experience is limited but I'm wondering if this could be tuned by having an optional list of `__name__` values where displayio updates would happen. I.e. if that was set to just `['__main__']` then `displayio` would only refresh in code within the main application. Is the feasible/useful?

tannewt, 2020-03-04:
Refreshing the screen in the background even though it is slow is a trade-off I'm willing to make because it makes things easier. We don't make any promises about the speed of the Python execution because of background work. I don't know of any code that assumes that Python is fast and breaks when it is slow.

We can address any specific cases where we need more speed separately. This case isn't an issue because it is expected.

closed 2020-03-04 by tannewt

kevinjwalters, 2020-03-04:
There are some examples where people are trying to do timed pulses with CP, e.g. https://learn.adafruit.com/controlling-a-classic-nintendo-r-o-b-robot-using-circuit-playground-express/programming but this is on CPX and there's no `displayio` involved.

tannewt, 2020-03-04:
@kevinjwalters thanks for pointing that case out! The recommended way of doing it is PulseOut rather than in a loop.

@TheKitty Would you like to update the example to PulseOut? Did you have reliability issues?

kevinjwalters, 2020-03-04:
I'm guilty of it too. I took a very simple library and did minimum modification for the lego vll protocol in https://learn.adafruit.com/bluetooth-remote-for-lego-droid/circuitpython but the minimum delay and granularity of that is 20ms...

kevinjwalters, 2020-03-07:
At the other end of the spectrum it's interesting to see there's already awareness that as CircuitPython starts running on faster processors then other assumptions can breakdown, e.g. switch debounce requirements: https://learn.adafruit.com/make-it-switch/debouncing#circuitpython-6-9

kevinjwalters, 2020-04-06:
To add to the general discussion around use of ms timing in CircuitPython, I just spotted a few more example of this in a library.

- https://github.com/adafruit/Adafruit_CircuitPython_SHT31D/blob/36379b2574a618814a9ff61297f234189afb6390/adafruit_sht31d.py#L187-L196
- https://github.com/adafruit/Adafruit_CircuitPython_SHT31D/blob/36379b2574a618814a9ff61297f234189afb6390/adafruit_sht31d.py#L207-L229

I'm not saying those will be affected by additional delays as I've not looked at what's going on there but I think it's worth keeping an eye on the area of i2c communications as I'm sure there's going to be a few "impatient" devices out there that behave unpredictably when delays of hundreds of milliseconds are introduced into the protocol.

kevinjwalters, 2020-05-07:
A recent very good guide article also needs precise timing to maintain the timing of its serial async protocol: https://learn.adafruit.com/clue-teletype-transmitter/code-the-tty-transmitter

tannewt, 2020-05-07:
> A recent very good guide article also needs precise timing to maintain the timing of its serial async protocol: https://learn.adafruit.com/clue-teletype-transmitter/code-the-tty-transmitter

This example isn't a reason to support precise timing from Python (which is impossible). It's a reason to expand PulseOut to support DAC output in addition to PWM carriers.

kevinjwalters, 2020-05-07:
> This example isn't a reason to support precise timing from Python (which is impossible). It's a reason to expand PulseOut to support DAC output in addition to PWM carriers.

Is there a ticket for that?

tannewt, 2020-05-08:
@kevinjwalters Not that I know of. Please open one. It'll be very low priority but good to track non-the-less.

kevinjwalters, 2020-05-28:
Another interesting case of timing in CircuitPython: [Question about stepper motor control
](https://forums.adafruit.com/viewtopic.php?f=50&t=165733)

A user is interested in achieving a certain rpm with a stepper motor and the discussion yields a suggestion which is likely to be interepreted as a while/for loop with 'kit1.stepper1.onestep(direction=stepper.FORWARD, style=stepper.INTERLEAVE); time.sleep(0.006)' in it.