adafruit/circuitpython #2673: remove size limit on displayio.Group

dhalbert, 2020-03-01:
`Group full` messages can be confusing. Could we make `Group` not need a size limit? It could be like a list, or it could resize itself as necessary.

See for example https://github.com/adafruit/Adafruit_CircuitPython_Display_Text/issues/11

dastels, 2020-03-01:
Would this lead to accidentally adding groups (which is what the limit guards against) and getting an even more confusing out of memory error?

jepler, 2020-03-01:
make group.size a settable property, possibly increase-only?  that would prevent "accidental" addition of groups, since you have to explicitly ask for the increase.  it would address the label use-case for not requiring the size to be specified upfront.

kmatch98, 2021-04-11:
Closed by: https://github.com/adafruit/circuitpython/pull/4233

closed 2021-04-11 by kmatch98