adafruit/circuitpython #2676: audio: continuous DMA from AudioIn, real-time audio filtering

jepler, 2020-03-03:
We'd like to be able to do real time audio filtering with CircuitPython.  Example: Read from mic, perform FIR filtering with ulab, output on speaker.

We would also like to process all data from the mic without interruption.  Example: Waterfall FFT demo

One option to enable this is to allow Python classes to be samples, to allow Python code to retrieve sample data, and to make microphones into audio samples.

Either M4 or nrf52840 would get the initial (PDM)AudioIn DMA work, as this is different per MCU.  nrf52840 because clue is the current hotness, M4 because its audio out is better for the filtering case.

A sketch of how this might look as a class outline is at https://gist.github.com/2701e024cdea2c2f60527d674a7b28bc

However, it might also turn out that this is infeasible due to GC, constraints on what we can do from background tasks etc., and instead we need to add a fixed-function pipeline of some sort.  In that scenario, a set of FIR taps could be associated with e.g., an AudioMixer object instead.

I'm initially associating this with the 5.x.0 features milestone because @ladyada mentioned this as a possible use case for ulab.  It may be too aggressive a timeline to have this done with the ulab release, so the milestone can be postponed as appropriate.

TheMindVirus, 2022-02-24:
Mozilla WebAudio API covers a large portion of what you would like to see working in CircuitPython `audioio`, `audiomixer` and `synthio`: https://developer.mozilla.org/en-US/docs/Web/API/Web_Audio_API
It is currently in Working-Draft for Javascript in Modern Web Browsers such as Firefox and Chrome and is very well documented, so much so that I have written several demo apps: https://github.com/adafruit/circuitpython/issues/5944

#### A CircuitPython port of this API would be able to bridge the gap between existing code in the web and microcontrollers.

While looking into porting some of the code that uses the Biquad Filter (potentially useful for the Equaliser for WinAmp),
I looked for a CircuitPython board that has a headphone jack. The Sony Spresense came to hand but required some setup.
I then noticed that the particular release of CircuitPython I had flashed to the board (7.0.0) was missing `audioio` and `audiomixer` entirely, along with an obvious pin definition for the headphone jack in `board`.

The next port of call was the newly updated Arduino Core (for which the SiLabs CP210x Serial Drivers had to be updated) which was able to upload sketches in bootloader mode with the microUSB on the mezzanine (with their firmware upload tool and some custom written batch files since the Arduino "burn bootloader" button appeared to be broken).

Spresense Upload Syntax: `flash_writer.exe -s -c %PORT% -b %BAUD% -d -n "./loader.espk"`
CircuitPython 7.0.0 Image: `flash_writer.exe -s -c COM10 -b 115200 -d -n "./adafruit-circuitpython-spresense-en_GB-7.0.0.spk"`

I wrote a simple starter sketch replicating a retro TV Beep that counts down before shows used to go on air.
The same sketch can be written for Raspberry Pi OS by muxing the headphone jack GPIO pins and writing directly to them,
but the Spresense makes it a lot more complicated than that by going through its own NuttX abstraction layer at runtime.

I think the Spresense board with some tweaks could be a really good DSP Accelerator for other CircuitPython boards,
especially those which don't have their own headphone jacks or other physical connectors for Audio/Video interfacing.
It could do this either by using its own SDK or running a separate instance of CircuitPython, in both cases completely isolated
from the board running your project thereby removing many compiler limitations found when running on one microcontroller.

More details about what I found trying to get the headphone jack working are included as comments in the Arduino Sketch attached below in `SpresenseTVBeep.ino`. I also decided to record and denoise the output captured directly from the board
after testing it with some well preserved Sony headphones.

Arduino Sketch and Audio Samples included: [SpresenseTVBeep.zip](https://github.com/adafruit/circuitpython/files/8129727/SpresenseTVBeep.zip)

tannewt, 2022-02-24:
@TheMindVirus I'd recommend adding an issue for supporting the existing audio apis on spresense too.

mfhepp, 2022-05-12:
I support this proposal; it would be really useful to be able to use e.g. the M4 CPU power to do FFT, digital filters etc. in a background process from CircuitPython without needing to go back to lower-level programming environments.

For example, people are doing Software-defined Radio (SDR) stuff on ATmega328Ps (see [here](https://dl2man.de/2-trusdx-assembly/)). So being able to at least do filtering of audio frequencies (low-pass, band-pass, high-pass), amplitude  modification, amplitude modulation, impuls shaping would be great.

From an architectural perspective and without having looked at available libraries, I think it would be sufficient to

1. be able to set up a DMA-based sampling-output cycle, basically a mapping from ADC to DAC port and setting a sampling rate, and
2. be able to register a "transfer function" from CircuitPython that translates the input sample to the output samples.

Does that sound realistic? And could we get this from "some distant future feature" to "near-term priorities"?

timchinowsky, 2024-07-07:
I've prototyped on rp2040 a set of minimal, non-breaking changes which enable processing of live sampled audio.  With these changes, the following code will sample audio from `A0` and play it out with PWM on `D12`:

```python
import analogbufio
import array
import audiocore
import audiopwmio
import board

buffer = array.array("H", [0x0000] * 8)
adc = analogbufio.BufferedIn(board.A0, sample_rate=10000)
adc.readinto(buffer, loop=True)
pwm = audiopwmio.PWMAudioOut(board.D12)
sample = audiocore.RawSample(buffer, sample_rate=10000, single_buffer=False)
pwm.play(sample)
```

Changes required:

1.  Add a `loop=True` option to `analogbufio.BufferedIn.readinto` so that instead of terminating `readinto` sets up looping DMA into `buffer`. 
2.  Add a `single_buffer=False` option to `audiocore.RawSample` so that the `RawSample` presents to `PWMAudioOut` as double-buffered instead of single-buffered.
3.  Tighten up the timing across the board so that everyone agrees on what 10000 Hz is.  For this to work without glitches timing must be exact, see #9396 and #9398, and right now I think there is also an issue with the use of the DMA pacer timer in `audiodma.c`, see https://forums.raspberrypi.com/viewtopic.php?t=373201.  To get this to work glitch-free I replaced the pacer with a PWM DREQ.

With this in place it should be straightforward to add features like multi-channel sampling using the ADC round-robin registers, and to implement DSP via functions which take in a buffer and return a processed buffer.    

Wondering where this might fit on the development roadmap, let me know if you'd like me to tidy it up and submit a PR.  `analogbufio` is not implemented on many ports but it does have a nice simplicity to it.

dhalbert, 2024-07-07:
@timchinowsky This is very nice! Would it also work with audiobusio?

timchinowsky, 2024-07-07:
It should work anywhere that uses the audio buffer protocol - basically it tricks a `RawSample` into providing live data.

jepler, 2024-07-07:
Does this idea pave the way to doing any processing on the audio data before it's output, such as applying a digital filter?

timchinowsky, 2024-07-07:
@jepler yes, by composing the `RawSample` with DSP functions that take in a buffer and return a buffer, so for instance instead of `play(sample)` you'd do `play(filter(sample))`, etc.  I'm planning to write a clipping function and a biquad filter to test this out.  Thinking the DSP functions will mostly modify input data in-place rather than require more allocation.  Adding, multiplying, etc. multiple signals should work too.

tannewt, 2024-07-08:
> Wondering where this might fit on the development roadmap, let me know if you'd like me to tidy it up and submit a PR.

I'm happy to do the reviews for this whenever you are ready. Folks would love to see this I'm sure.

timchinowsky, 2024-07-17:
Right now in rp2040 audio DMA in `audiopwmio` always proceeds at a rate set by a DMA timer.  I've found these to be not precise enough to synchronize with e.g. the ADC pacing timer.  It would be helpful to be able to specify a sample rate by reference to another peripheral, i.e. "use whatever sample rate the ADC pacer is set to."   One way this could be done is to define some high sample rates as special-purpose codes, e.g. a sample rate of 0xFFFFFF00 would be interpreted as "use the ADC DREQ", etc.  Thoughts?

timchinowsky, 2024-08-02:
Signal quality is never going to be good with the RP2040 ADC so I'm shifting focus to an I2S audio codec, [TAC5212](https://www.ti.com/product/TAC5212).  I've got the ADC and DAC working and am working on implementing `I2SIn` a la #5456. 
[image]

ladyada, 2024-08-02:
thats a cute I2S codec, i ordered some samples of the TAC5242

timchinowsky, 2024-08-02:
I've got ~5 populated breakout boards for the TAC5212 that are up for grabs, if anyone's interested in messing with it.

tannewt, 2024-08-05:
> It would be helpful to be able to specify a sample rate by reference to another peripheral, i.e. "use whatever sample rate the ADC pacer is set to." One way this could be done is to define some high [...]

Would it be enough for us to coordinate internally so that the ADC and PWM output dma's use the same pacing timer when set to the same rate? I don't really like the special value. Instead, I'd just pass the object you want to tie to.

timchinowsky, 2024-08-05:
That would be better, it's just more code.

timchinowsky, 2024-08-05:
Re: the TAC5212, these things are great, up to 768 kHz sampling at up to 32 bits, and you can daisy-chain the devices to get up to 8 channels over one pin.  To do them justice, planning to use `rp2pio` with whatever modifications are needed to support continuous streaming from input to output.  That module is kind of the ultimate as far as commanding low-level hardware performance with high-level CP functions.  There is a lot of variety in how codecs like to be spoken to, and it will be nice to surface that in the CP code.

tannewt, 2024-08-06:
> To do them justice, planning to use `rp2pio` with whatever modifications are needed to support continuous streaming from input to output.

Note that this isn't portable. An I2SIn would be better because it could be implemented on other platforms. RP2 can use PIO internally.

> That would be better, it's just more code. Using a magic number is a ~two-line change.

More code is fine. Magic numbers are too much magic imo.

timchinowsky, 2024-08-06:
> Note that this isn't portable. An I2SIn would be better because it could be implemented on other platforms. RP2 can use PIO internally.

Yeah, I was thinking to do both, like right now there's both https://github.com/adafruit/Adafruit_CircuitPython_PIOASM/blob/main/examples/pioasm_i2sout.py and the same pio code in https://github.com/adafruit/circuitpython/blob/main/ports/raspberrypi/common-hal/audiobusio/I2SOut.c.

timchinowsky, 2024-08-06:
Although I think what we're really going to want is an `I2SInOut` that does both directions with the same clocks.

timchinowsky, 2024-08-28:
Progress with the TAC5xxx codecs using `rp2pio` posted here: https://github.com/timchinowsky/tac5.
Next is (1) add a background read to `rp2pio` to complement the background write which is already there and (2) ponder how to connect the reads to the writes with some CP signal processing, integrating with existing `audiocore` paradigms where possible.

timchinowsky, 2024-09-24:
Added draft `background_read` method: https://github.com/adafruit/circuitpython/pull/9659.  Seems to work with TAC5 codec, needs more testing.

timchinowsky, 2024-11-12:
I've gotten pretty far along on this in #9659, just added some effects demos (ring modulator, echo).  My approach is to do DAC/ADC I/O in the background, but manipulate the double buffers in the foreground.  I like how this makes the process more modular and transparent.  Here is a snippet showing how an audio file is read from SD, processed, and sent to the codec:
```python
with open(filename, 'r') as f:
                while repeat or n == -1:
                    while n != 0:
                        b = self.pcm.pio.last_write
                        if len(b) > 0:
                            status.value = True
                            n = f.readinto(b)
                            status.value = False
                            status.value = True
                            if process is not None:
                                self.pcm.pio.process(b, parameters=process)
                            status.value = False
                    if repeat:
                        print('repeating...')
                        f.seek(0)
                        n = -1
```
the `last_write` property returns the buffer which was most recently sent to the codec (and which needs to be refilled).  The main loop fills the buffer with `f.readinto(b)`, processes it with the method `pio.process`, then waits for the next buffer.  Toggling of the `status` pin provides timing info.   For example, for playback of an 8-channel, 16-bit, 8000 Hz sample, with a 2048-sample echo applied, the command is `t.play('/count_8ch_16bit_8000Hz.raw', process=tac5.echo(length=2048))` and the timing looks like this:

[image]

Here the buffers contain 400 samples of each channel, so the 800 samples of both buffers are output in 100 ms, as expected for 8000 Hz sampling.   Seems to run just fine on the RP2350.  The processing is faster there, while the I/O, presumably SPI-limited, is similar for RP2350 and RP2040.

Here's the C function which does the processing, which I put in `rp2pio` for now, but could and should be somewhere else.  Its job is to modify the contents of a `target` buffer in a way that specified by a `parameters` buffer.  The data in `parameters` describes the type of processing to be done, any parameters required by the processing (e.g. filter coefficients).  Any scratch variables (e.g. filter memory) that needs to persist between buffers is also stored in `parameters`.  

```c
bool common_hal_rp2pio_statemachine_process(rp2pio_statemachine_obj_t *self, uint8_t stride_in_bytes,
    sm_buf_info *target_buf_info, sm_buf_info *parameters_buf_info, sm_buf_info *input_buf_info) {
    if (parameters_buf_info->info.len > 2) {
        if (stride_in_bytes == 2) {
            int16_t op = ((int16_t *)parameters_buf_info->info.buf)[0];
            int16_t *target;
            target = (int16_t *)target_buf_info->info.buf;

            switch (op) {
                case 1: // multiply
                    int16_t *factor;
                    uint16_t *factor_index;
                    size_t factor_samples;
                    factor_samples = (parameters_buf_info->info.len - (2 * stride_in_bytes)) / stride_in_bytes;
                    factor_index = ((uint16_t *)(parameters_buf_info->info.buf)) + 1;
                    factor = ((int16_t *)(parameters_buf_info->info.buf)) + 2;
                    for (size_t i = 0; i < (target_buf_info->info.len / stride_in_bytes); i++) {
                        target[i] = ((int32_t)target[i] * (int32_t)factor[*factor_index]) >> 16;
                        (*factor_index)++;
                        if (*factor_index >= factor_samples) {
                            *factor_index = 0;
                        }
                    }
                    break;

                case 2: // echo
                    int16_t *memory;
                    uint16_t *memory_index;
                    size_t memory_samples;
                    memory_samples = (parameters_buf_info->info.len - (2 * stride_in_bytes)) / stride_in_bytes;
                    memory_index = ((uint16_t *)(parameters_buf_info->info.buf)) + 1;
                    memory = ((int16_t *)(parameters_buf_info->info.buf)) + 2;
                    for (size_t i = 0; i < (target_buf_info->info.len / stride_in_bytes); i++) {
                        target[i] = (target[i] + (memory[*memory_index] >> 1)) >> 1; // fixed 50% decay for now
                        memory[*memory_index] = target[i];
                        (*memory_index)++;
                        if (*memory_index >= memory_samples) {
                            *memory_index = 0;
                        }
                    }
                    break;
            }
        }
    }
    return 1;
}
```
I was thinking that `ulab` would be useful here, as is mentioned in the opening comment of this issue, but so far not really.  `ulab` does not know how to do fixed-point arithmetic (e.g. Q1.15), and is also limited in its ability to calculate results in-place rather than create a new array.

I've been focusing on codec I/O using `rp2pio`, but this approach could be generalized to other interfaces by providing them with similar `background_read`/`background_write` methods and `last_read`/`last_write` properties.  For the processing, a library specifically designed for fast in-place modification of arrays would be handy, whether stand-alone or as an extension to `ulab`.