adafruit/circuitpython #2679: STM32F4 Discovery: Unable to set port D pin direction as OUTPUT

louismdavis, 2020-03-03:
I have been testing out 5.0.0 on an STM32F407 Discovery board and noticed I was unable to set any of the LED pins as OUTPUT. After setting the pin direction, it would always read back as INPUT.

I was looking through the source and noticed within the stm32f4_peripherals_gpio_init() function that only ports A,B and C are enabled.

ports/stm32f4/peripherals/stm32f4/stm32f407xx/gpio.c
 
void stm32f4_peripherals_gpio_init(void) {
    //Enable all GPIO for now
    __HAL_RCC_GPIOC_CLK_ENABLE();
    __HAL_RCC_GPIOA_CLK_ENABLE();
    __HAL_RCC_GPIOB_CLK_ENABLE();

tannewt, 2020-03-03:
Thanks for the issue @louismdavis !

hierophect, 2020-03-03:
Dang, thanks for spotting this! This was a testing oversight on my part, didn't go through enough pins! Should get this done today.

PR #2680 (2020-03-04): STM32: Fix restriction on GPIO clock enables for F407 Discovery board
Also expanded for F405 SoC definition, for the possible future case a F405 board with the expanded pins is added. Resolves #2679.
closed 2020-03-04 by tannewt