adafruit/circuitpython #2687: object has different identities during and after import

dhalbert, 2020-03-07:
nested.py:
```python
l = [[]]
l0 = l[0]

print("l[0]", id(l[0]))
print(l0, id(l0))
```
```
Adafruit CircuitPython 3.1.2 on 2019-01-07; Adafruit Metro M0 Express with samd21g18
>>> from nested import *
l[0] 536876704
[] 536876704
>>> id(l[0])
536876704
>>> id(l0)
536896144   # should not be different from id(l[0])
>>> 
```
In CircuitPython 2.x, MicroPython, and CPython the `id()` values are the same in the REPL. My guess is that this has to do with making an object long-lived during compilation, since this shows up in 3.x.

Thanks @TG-Techie for spotting this. Tagging @tannewt.

tannewt, 2020-03-09:
I think the correct thing would be to have the later id of `l[0]` be 536896144. I don't mind that the ids themselves change after import but we should create two separate objects.

I do agree that this is a long lived issue because we assume that only one reference of module variable exist. (We recurse from the module dict down and don't look elsewhere for references.)

I don't think this is urgent and would argue for it to be "Long Term" because there is a simple and easy workaround: Just assign `l0 = l[0]` after the import rather than in the module.

TG-Techie, 2020-03-11:
@tannewt I agree it should be ling term. I think very few people will find this bug and the temporary fix is quite easy.

It could be that I am miss-interpreting "the correct thing would be to have the later id of l[0] be 536896144" but I think that the object shouldn't be made into a separate copy after import. I think that would be perfectly acceptable for immutable objects like `int`s, `str`s,` tuple`s, etc but mutable objects should not be duplicated. I think when the list is edited in the repl it should change in the module.

 to me, this seems intuitively correct because any import in python is really just syntactic sugar for the __import__ function for `getattr`ing objects from inside it (with the `.`). thus if the same module is imported into two other python files they (I believe) should both be getting references to the same list object b/c the lists is within the same module object. 
like how `from time import sleep` is equivalent to `sleep = __import__('time').sleep`

TG-Techie, 2020-03-11:
meta importers and other CPython extensibilities aside

tannewt, 2020-03-11:
> @tannewt I agree it should be ling term. I think very few people will find this bug and the temporary fix is quite easy.
> 
> It could be that I am miss-interpreting "the correct thing would be to have the later id of l[0] be 536896144":
> I think that the object shouldn't be made into a separate copy after import. I think that would be perfectly acceptable for immutable objects like `int`s, `str`s,` tuple`s, etc but mutable objects should not be duplicated. I think when the list is edited in the repl it should change in the module.

It's not meant to be a copy (that is the bug). It is meant to be a move. The id will change but it will change consistently between the two objects. If you add a function that will print the id from within the module's scope then the new id should be there as well. The object is moved at the end of the import and the existing prints are before that.

TG-Techie, 2020-03-11:
Ah, I did misinterpret. Sounds perfect

jepler, 2023-06-22:
Restating this without use of `id`:

Put the following in `nested.py`:
```py
# content of nested.py
l = [[]]
l0 = l[0]
```

and the following in `code.py`:
```py
# runtime
import nested
nested.l0.append("test")
print(nested.l)
```

With python3, it prints `[['test']]`, while with circuitpython it prints `[[]]``, because in circuitpython `nested.l0` and `nested.l[0]` are not the same object due to the implementation of "long lived" being unable to move objects.

dhalbert, 2023-10-27:
Fixed by #8281, which removed long-lived allocations, and so removed objects moving around after creation.

closed 2023-10-27 by dhalbert