adafruit/circuitpython #2696: new functionality for displayio refresh() to support a simple, immediate refresh

kevinjwalters, 2020-03-11:
For those not using the `auto_refresh` mode, the current `refresh()` in `displayio` is tailored toward applications which would like to achieve a constant frame rate with a feature to drop updates under certain conditions. The current algorithm/options are described in detail in https://github.com/adafruit/circuitpython/issues/2310#issuecomment-596834307

For the use case of applications which want to manipulate displayio objects and then do a manual refresh at the end of those updates which is immediate and returns without additional sleeps there's no easy way to use `refresh()` at the moment to achieve this.

kmatch98, 2020-08-30:
@kevinjwalters and @tannewt, I’m willing to pitch in on this issue. I read through the related issue and here’s what I gather:

Need a way to call for an immediate (blocking?) screen refresh.  I assume this would be an empty call `display.refresh()` or with `target_frames_per_second=None`.

I would suggest that this include an option to treat the whole screen as `dirty` and do a full redraw of all items.  This could be an interim solution for structures that do not have dirty rectangle tracking implemented (see issue for Shape https://github.com/adafruit/circuitpython/issues/3201). 

If you can point me in the right direction about which files to look, I will try my best to contribute.  In particular, I’d ask for your help to understand how:
- displayio handles the different structures dirty rectangle tracking (I see one structure called `full_refresh` in function `displayio_display_core_show` but I’m not sure what decisions are based on that value.)
- what are the key functions and data structures that trigger a screen refresh. (I see a variable named `first_manual_refresh` that sounds like a possible avenue to try but I can’t follow what uses this.)

Note: Generally I have difficulty tracing all these connections. Do you have suggestions on ways of searching the different files for code to learn how it all fits?

tannewt, 2020-08-31:
> @kevinjwalters and @tannewt, I’m willing to pitch in on this issue. I read through the related issue and here’s what I gather:

Thank you! I'm happy to give pointers on this.

> Need a way to call for an immediate (blocking?) screen refresh. I assume this would be an empty call `display.refresh()` or with `target_frames_per_second=None`.

Yup, blocking is good. I think we could change the API so that the default means now. It'll be good to have the target fps explicit otherwise.

> I would suggest that this include an option to treat the whole screen as `dirty` and do a full redraw of all items. This could be an interim solution for structures that do not have dirty rectangle tracking implemented (see issue for Shape #3201).

I'd rather not do this because it's working around a bug that should be fixed.

> If you can point me in the right direction about which files to look, I will try my best to contribute. In particular, I’d ask for your help to understand how:
>
>     * displayio handles the different [...]

This is used to trigger a full refresh when the top level group is changed.

>     * what are the key functions and data structures that trigger a screen refresh. (I see a variable named `first_manual_refresh` that sounds like a possible avenue to try but I can’t follow what uses this.)

[`common_hal_displayio_display_refresh`](https://github.com/adafruit/circuitpython/blob/main/shared-module/displayio/Display.c#L354) is the top C function called for SPI devices. The different display types have similar functions. This top level is where the timing is done.

I think what you'll want to do is make None mean `target_ms_per_frame` is `0xffffffff` and then skip the top level `if` in that case.

> 
>     * There is a variable `core.full_refresh` that impacts `_get_refresh_areas` (called by `_refresh_display`)
> 
>     * I found a function `_refresh_display` which looks like it may be the key, located in `shared-modules/Display.c`.  If this is combined with `core.full_refresh`, will that trigger a full update of the screen?

This is one level lower than you want I think because this is called when we've already decided its time to refresh. It manages *what* to refresh, if anything. First, it gets all changed rectangles from the top-level group and its layers. Then, it iterates through them all to update those portions of the screen. If there are no areas, it still counts as a frame for timing.

> Note: Generally I have difficulty tracing all these connections. Do you have suggestions on ways of searching the different files for code to learn how it all fits?

It's complex code so don't feel bad that it's tough to follow. The easiest way is to probably just ask me or someone else early. I'm happy to give a high level overview of how something works any time you need.

kevinjwalters, 2020-09-02:
Is there a (significant, single) dma portion? If so, an option to return while that is running would be useful?

deshipu, 2020-09-02:
I experimented with this a while ago, when `displayio` was just being created, and the gains were negligible.

closed 2020-09-02 by tannewt