adafruit/circuitpython #2700: board.SPI() hangs after soft reboot

jerryneedell, 2020-03-12:
When using my rfm69 on a feather_m4_express or teensy4.0 I find that the accessing SPI hangs after a soft reboot if I create the SPI interface with board.SPI() rather than busio.SPI(board.SCK.......).   After a power cycle, either method works. I am using the current master commit d988af0 ...

The hang is apprarently happening here:
https://github.com/adafruit/Adafruit_CircuitPython_BusDevice/blob/master/adafruit_bus_device/spi_device.py#L81

Is there is differece in how the lock is handled with board.SPI() and bus.SPI((....)?

hierophect, 2020-09-16:
Is this still happening for you? I might want to test it on STM32 and ESP32, see if it occurs on those too.

jerryneedell, 2020-09-16:
This appears to be resolved - as well as being duplicated -- see #3266

jerryneedell, 2020-09-16:
resolved.

closed 2020-09-16 by jerryneedell