adafruit/circuitpython #2702: build .uf2 for meowbit

dhalbert, 2020-03-13:
STM-based `meowbit_v121` has a uf2 bootloader. Build and upload a .uf2 for the board, not a `.bin.

Thanks  pcoxall aka @Mr-coxall from https://forums.adafruit.com/viewtopic.php?f=60&t=163432.

hierophect, 2020-04-17:
@dhalbert anything I need to do here or is this a website issue?

dhalbert, 2020-04-17:
We need to change `tools/build_board_info.py` so that it uploads the uf2 for the board. I.e. Add
```
    "meowbit_v121": UF2,
```
to `extension_by_board`. That will upload the .uf2 instead of the .bin.

Mr-Coxall, 2020-06-17:
I just checked the website:
https://circuitpython.org/board/meowbit_v121/
It still does not have a .UF2 file to download.

FoamyGuy, 2020-06-17:
@Mr-Coxall I think it may show up on the downloads page when there is a new release made. If I understand correctly the change has been merged into the repo but there hasn't been a new release of circuitpython yet since then so it's not visibly on the download page yet.

Mr-Coxall, 2020-06-27:
So a new release has come and there is still just a *.bin file and no *.uf2 file.
Any ideas?

hierophect, 2020-06-29:
@dhalbert is there something else I need to do here to get this up on the website? It should have gone up with the release.

dhalbert, 2020-06-29:
We didn't make a new release yet, we just tagged 6.0.0-alpha.0, but there was no release.

.uf2 artifacts for the meowbit started showing up as of June 17: https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/meowbit_v121/en_US/

But there hasn't been a commit to https://github.com/adafruit/circuitpython-org/blob/master/_data/files.json since June 8, when the last 5.4.0 beta was released.

I'll check during the meeting today whether 6.0.0-something is imminent. If not we can fix this by hand, but we need to add a 5.3.0 .uf2 build to Amazon S3.

Mr-Coxall, 2020-06-29:
OK, thanks

tannewt, 2020-06-30:
I plan on doing a 6.0.0 pre-release this week. (Probably Thursday or Friday.)

tannewt, 2020-07-13:
This is done with 6.0.0-alpha.1

Mr-Coxall, 2020-07-15:
When I place the MeowBit in boot mode to load a*.uf2 file and then load 6.0.0-alpha.1, it seemingly does load on the MeowBit, but then it just returns to the loading screen, not circuitpython. When I load a *.uf2 file from Makecode, it does load and runs the code.
Here is the "Hello, World!" Makecode file I am using that does work.
 
[arcade-Hello-World.uf2.zip](https://github.com/adafruit/circuitpython/files/4926047/arcade-Hello-World.uf2.zip)

hierophect, 2020-07-15:
From what I remember you need to reset the meowbit for it to enter circuitpython. It cycles between the two modes.

hierophect, 2020-07-15:
I'll have to load it up again and see what the deal is, in that case. Haven't tried the Meowbit in a while, there may have been a regression.

PR #3039 (2020-06-17): STM32: make Meowbit default to UF2 on website
Resolves #2702