adafruit/circuitpython #2704: List of supported SPIFlash

NightSkySK, 2020-03-13:
Searching guidelines for builds "How to Add a New Board to CircuitPython" and other sources I could not find any information more than "We only guarantee support for the flash chips used on Adafruit boards" 

I would suggest creating a list of supported SPIFlash which are supported by CircuitPython, this would help makers to ensure that the component which they are going to use is actually supported.

siddacious, 2020-03-13:
@NightSkySK I think this is certainly a good idea and would be helpful to have. In the meanwhile you can find a list of currently supported chips in this file:
https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/external_flash/devices.h

It may be a bit terse but the info you need is there.

I think it would be a good approach to have the list you suggested generated from that file

dhalbert, 2020-03-13:
We can provide a cross-reference to that file from the Learn Guide, since the file is the "ground truth" on what is supported.

bwshockley, 2020-03-13:
@NightSkySK - you can also add a new flash to the list - I did this before, if you find one you'd like to use.  You'll want to test it out first.  They pulled in my change request to support my board into the linked devices.h file.

NightSkySK, 2020-03-13:
@bwshockley Lucky me, my W25Q32JVSSIQ is on the list as W25Q32JV_IQ, I've just struggled to find any place where I can find the list of all supported devices. I've searched other board definitions to find if any other has the same flashSPI.

dhalbert, 2020-03-14:
I've added information about looking in `supervisor/shared/external-flash/devics.h` to https://learn.adafruit.com/how-to-add-a-new-board-to-circuitpython/initial-configuration. I think that should clarify things.

closed 2020-03-14 by dhalbert