adafruit/circuitpython #2707: RGB_LED_GREEN in use error on Particle Xenon

jerryneedell, 2020-03-15:
Attempting to access the RGB_GREEN_LED on a Xenon throws this error:
```Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-87-g3a5f79ace on 2020-03-13; Particle Xenon with nRF52840
>>> import time
>>> import board
>>> from digitalio import DigitalInOut, Direction, Pull
>>> led = DigitalInOut(board.RGB_LED_GREEN)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: RGB_LED_GREEN in use
>>> ```

The same code executes normally on an Argon
```
Auto-reload is on. Simply save files over USB to run them or enter REPL to disable.

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.0.0-87-g3a5f79ace on 2020-03-13; Particle Argon with nRF52840
>>> import time
>>> import board
>>> from digitalio import DigitalInOut, Direction, Pull
>>> led = DigitalInOut(board.RGB_LED_GREEN)
>>> 

```

jepler, 2020-03-15:
I notice that xenon has
```
#define CP_RGB_STATUS_R        (&pin_P0_13)
```
while argon/boron have a macro called MICROPY_HW_RGB_LED_RED.  Only the former is used.  I think this means that there are two bugs: first that boards that use CP_RGB_STATUS_LED don't cooperate with DigitalInOut in the way that neopixels do (this bug, affecting Xenon); second, that the status LED doesn't operate on Argon/Boron.

@jerryneedell can you confirm whether or not the status LED works normally on Argon/Boron while in CircuitPython?

axiomatic-io, 2020-03-15:
I think I found the issue but I don't understand what the purpose is.

@tannewt in commit 5610e0 there is this:

`#if defined(CP_RGB_STATUS_LED)
        // TODO: Support sharing status LED with user.
    #endif`

It looks like before that commit the pins underlying the status LED are reset in a similar manner to the Argon and Boron. Can you explain to me the thoughts behind making the Xenon have a different implementation of the status LED?

jerryneedell, 2020-03-15:
@jepler On my Argon the Status LED is not lit at all in CP.  IT is red briefly after RESET the off. I can set it manually.
on the Xenon the Led is  red briefly after RESET then white all the time.

In the bootloader - the XENON LED is Green - same for the ARGON

jepler, 2020-03-15:
My interpretation is:  Argon/Boron don't have the defines that create the PWM'd RGB status LED.  This leaves the pins able to be used for other purposes.  The Xenon *does* have the defines, but because of the TODO you can't take over the pins for your own use.  However, it has an additional bug which causes the LED to show white instead of the normal CP status colors.  Three ... three bugs.

If you didn't mind making your own build, you can simply remove the CP_RGB_STATUS_R, G, and B in ports/nrf/boards/particle_xenon/mpconfigboard.h to get control of the RGB LED from software.

jerryneedell, 2020-03-15:
@jepler Thanks for tracking this down. I posted the issue in response to a discord conversation where this was reported. I don't have an immediate need for it. 
Should the Xenon be changed to have the same configuration as the Argon/Boron? I'll be happy to look into that and put in a PR.  
Or should this wait for the full implementation of the status LED.

jepler, 2020-03-16:
@jerryneedell Since it appears the status LED doesn't work properly on any board in the family, I am not bothered by the idea of Xenon to match Argon/Boron so the LED and let it be controlled by user code.  If someone wants to contribute a proper fix that makes the status LED work, and also makes the user use of the pins work down the line, it doesn't create much additional work from what I see.

axiomatic-io, 2020-03-16:
@jerryneedell What exactly are the functional specs of the status LED supposed to be?

jerryneedell, 2020-03-16:
> @jerryneedell What exactly are the functional specs of the status LED supposed to be?

 I think it is all controlled here https://github.com/adafruit/circuitpython/blob/master/supervisor/shared/rgb_led_status.c  but I would check with @tannewt or @dhalbert.

tannewt, 2020-07-14:
Tricolor status LEDs should use three PWMOuts to produce the status color. When a user attempts to use any of the three pins, the PWM should be deinit to free the pin for the user. After the pin is deinit by user code, the RGB status LED should reinitialize it and continue to use it.

ScruffR, 2021-05-18:
Has there been any progress on this front?
In 6.2.0 I still cannot find a way to use the oboard RGB LED on the Xenon.

@tannewt, if I want to deinit the pin I'd need to know which object to execute the method on (e.g.  `status_rgb_led_r.deinit()`).
But how would I find out which are the correct objects to use?

tannewt, 2021-05-18:
@ScruffR I think it'll be fixed by https://github.com/adafruit/circuitpython/pull/4743 because it doesn't use the status led during user code execution anymore.

ScruffR, 2021-05-19:
Thanks, and good to know :+1: 
I'll be waiting for 7.0.0 then as I'm not desperately needing the onboard RGB LED right now. Enough other stuff to do first :blush:

jrbski, 2021-06-06:
> Thanks, and good to know 👍
> I'll be waiting for 7.0.0 then as I'm not desperately needing the onboard RGB LED right now. Enough other stuff to do first 😊

I just tested circuitpython 7 alpha3, and was able to successfully control the RGB on XENON using the adafruit_rgbled library. Code snippet below just sets it to blue, but I was also able to cycle it R->G->B->R... also. It does look like #4743 does correct the issue.

`pixel_pin = board.RGB_LED_RED
pixel_pin2 = board.RGB_LED_GREEN
pixel_pin3 = board.RGB_LED_BLUE

led = adafruit_rgbled.RGBLED(pixel_pin, pixel_pin2, pixel_pin3, invert_pwm = True)
led.color = (0, 0, 255)
`

jepler, 2021-06-07:
Thank you for testing!

closed 2021-06-07 by jepler