adafruit/circuitpython #2729: General Library Idea: We should move towards having library functions return error conditions to the caller.

geekguy-wy, 2020-03-25:
IDEA: On library functions that raise an error condition, we should move towards having the error condition returned to the caller if it can be represented as a boolean or an int. We have already done this with the send() function in the adafruit_rfm69 library and I am using this in my current code. It works very well!

Personally, I would much rather handle errors in my code than have an error raised and have my code crash. It is better for the caller to be able to handle errors as much as possible.
For an int, zero would mean success and any other value is some type of error. This allows multiple error conditions to be documented and returned to the caller. For booleans, it would be a simple True means success and False means the operation failed.

dastels, 2020-03-25:
That's fine in simple cases, or where the function doesn't have a legitimate value to return.  But e.g. a function that return a boolean... how do you indicate an error? If any int is a valid result?

Go uses its multiple return value capability to have the convention of returning a separate error value (nil means no error). That approach could be used in Python as well. But Python has exception handling for this (Go doesn't).

My feeling is that CircuitPython should use the established Python idioms and approaches.

See: [Error Handling and Go](https://blog.golang.org/error-handling-and-go) and [Why Go Gets Exceptions Right](https://dave.cheney.net/2012/01/18/why-go-gets-exceptions-right)

geekguy-wy, 2020-03-25:
In some cases that return ints as valid values, you can return None for failure, or just return None for all failures represented as booleans. If it is not None then the operation succeeded. For cases where ints are valid return values, those values are _usually_ in a range. You can return a value _outside_ the acceptable range of return values. If the int values are all positive, return a negative error code. If the acceptable ints are all negative, return a positive error code.

I use these standards in my own code all the time and it all works very well. So far, I have not found any case where this does not work.

geekguy-wy, 2020-03-25:
One last thing for now. Just because there are established idioms to handle a lot of things, it does not mean there is no room for a new idiom or idea. Go is not Python. We are using Python, so we must work within its boundaries.

ladyada, 2020-03-25:
@geekguy-wy you can easily make multiple kinds of exception class names, throwing exceptions on error is the best way to indicate something went wrong and its up to the caller to catch the exception.

geekguy-wy, 2020-03-25:
@ladyada I understand that. However, I still believe _very strongly_ that errors should not be tripped from library routines. Library routines should be _quiet_ and not disturb the countryside. ;) It should not require try..except..finally clauses just to catch errors tripped by a library routine. It is very easy to make library routines quiet. :)

ladyada, 2020-03-25:
i think C/C++ users agree...the goal of python exceptions is to not let people skip or forget those errors, and to error as soon as possilbe!

dandelany, 2020-03-25:
As someone who writes a decent amount of python for work (though not CircuitPython, yet), just wanted to chime in in support of exceptions. Yes, they can be disruptive when they kill your program, but the philosophy/python idiomatic "way" is that accidentally swallowed/silent errors are much worse.

> Library routines should be quiet and not disturb the countryside.

Look to the Python standard library for guidance here - they are library routines too:

```python
>>> abs("hi")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: bad operand type for abs(): 'str'
```

Returning `None` in ^this case, or an error string, would be a decidedly worse experience IMO.

geekguy-wy, 2020-03-25:
Try...Except..Finally blocks increase code size. There is no reason to do this, either in a library routine or mainline script code. When return values and parameters are properly documented, there are no surprises, and no code expansion due to try..except..finally blocks. Again, I say that just because something is done a certain way, it does not mean it has to continue to be done that same way. It is fine for the Python standard libraries to do things the way they do, and I have to deal with it whether I like it or not. That does _**not**_ mean everything else has to be done the same way.

“The most damaging phrase in the language is ‘We’ve always done it this way’.”-- Grace Hopper

dhalbert, 2020-03-26:
It is a deep convention in Python to use exceptions to indicate errors. From [PEP 20: The Zen of Python:](https://www.python.org/dev/peps/pep-0020/) _Errors should never pass silently._ For more details: https://inventwithpython.com/blog/2018/08/17/the-zen-of-python-explained/ (and many more such explanations). You can argue with it, but we try to be Pythonic in general, so I think you'll have to live with it.

dandelany, 2020-03-26:
^ & respectfully, it's not just done "because we've always done it this way" - it's because a lot of smart people have put a lot of thought into the problem over a long period of time & we happen to agree with the conclusion they reached. It's certainly a fair thing to disagree about, but please don't assume that it's based on dogma.

ladyada, 2020-03-26:
hey folks, this discussion is getting a little off topic for circuitpython. we follow python conventions. some folks like some, some dont. hey, i am not a fan of whitespace-effect :D

closed 2020-03-26 by ladyada

tannewt, 2020-03-26:
I just wanted to chime in and say that often times an exception is generated due to invalid inputs to a function call. When this occurs a better fix than try/except is to use `if` statements to validate soon-to-be inputs before making the call. This makes the logic on handling those corner cases clearer. This isn't the case when network or device errors occur though.