adafruit/circuitpython #2738: Add ability to use SD card as main storage

geekguy-wy, 2020-03-29:
I know Micropython has this ability because I have a 16Gb SD card in my PyBoard 1.1.

It would be _**really nice**_ to be able to use an SD card for the main storage instead of flash or spi flash. This might breathe new life into some M0 boards that are not really usable for Circuitpython right now because there is not enough main storage available after loading Circuitpython.

tannewt, 2020-03-30:
I'd rather not remove the internal flash filesystem for non-Express boards because we then risk not having a block device to show as CIRCUITPY when an SD card is not present. We could do a better job supporting libraries loaded from SD card though.

Am I interpreting your suggestion right? Removing internal flash and requiring an SD card.

geekguy-wy, 2020-03-30:
Oh, no, no. I do not suggest removing the flash in favor of the SD card. I just want to add the SD card as an alternative to flash for loading scripts and libraries. For instance, I have two Feather M0 RFM69 boards that I would really like to use with Circuitpython, but there is not enough flash left after loading Circuitpython. I would like to be able to add either a wired up SD card breakout or an Adalogger FeatherWing so I could load scripts and libraries from the SD card.

tannewt, 2020-03-31:
So the internal flash filesystem is too small for you? You should be able to mount an SD card using a breakout already: https://github.com/adafruit/Adafruit_CircuitPython_SD/blob/master/examples/sd_read_simpletest.py

(It does take memory itself though.)

geekguy-wy, 2020-03-31:
Yes. On a Feather M0 RFM69, I have 26.1 kb of storage left with Circuitpython 4.x.x loaded. I have not updated those boards to 5.x.x yet because they are really not usable with Circuitpython at this time. I realize the M0 also has a memory limit with just 32 kb RAM too, but the flash is just not enough storage.

makermelissa, 2020-03-31:
I think being able to use the SD card to run your `code.py` off of would be handy even for an M4 if you wanted to change between programs easily.

dhalbert, 2020-03-31:
It is already possible to import code on the sd card. It would be something like this:

```python
import busio
import digitalio
import board
import storage
import sys
import adafruit_sdcard

SD_CS = board.SD_CS  # setup for M0 Adalogger; change as needed

# Connect to the card and mount the filesystem.
spi = board.SPI()
cs = digitalio.DigitalInOut(SD_CS)
sdcard = adafruit_sdcard.SDCard(spi, cs)
vfs = storage.VfsFat(sdcard)
storage.mount(vfs, "/sd")

# add /sd to the end of sys.path. You could add it anywhere.
sys.path.append("/sd")

import something_on_the_sd_card
```

makermelissa, 2020-03-31:
Excellent

geekguy-wy, 2020-03-31:
Oh, OK! I was not aware that we could do that. There is no need for this issue then, so I will close it now.

closed 2020-03-31 by geekguy-wy