adafruit/circuitpython #2745: Travis builds occasionally erroneously fail

xobs, 2020-04-02:
For some reason, Travis CI does not seem reliable.  It seems to fail about 5% of the time during the `Install deps` phase:

```
  Could not connect to azure.archive.ubuntu.com:80 (52.177.174.250), connection timed out
Err:2 http://security.ubuntu.com/ubuntu xenial-security/main amd64 gettext amd64 0.19.7-2ubuntu3.1
  Unable to connect to azure.archive.ubuntu.com:http:
Ign:3 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libasprintf-dev amd64 0.19.7-2ubuntu3.1
Ign:4 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libgettextpo0 amd64 0.19.7-2ubuntu3.1
Ign:5 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libgettextpo-dev amd64 0.19.7-2ubuntu3.1
Get:2 http://security.ubuntu.com/ubuntu xenial-security/main amd64 gettext amd64 0.19.7-2ubuntu3.1 [1,082 kB]
Get:3 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libasprintf-dev amd64 0.19.7-2ubuntu3.1 [4,778 B]
Get:4 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libgettextpo0 amd64 0.19.7-2ubuntu3.1 [125 kB]
Get:5 http://security.ubuntu.com/ubuntu xenial-security/main amd64 libgettextpo-dev amd64 0.19.7-2ubuntu3.1 [141 kB]
E: Failed to fetch http://azure.archive.ubuntu.com/ubuntu/pool/main/libu/libunistring/libunistring0_0.9.3-5.2ubuntu1_amd64.deb  Could not connect to azure.archive.ubuntu.com:80 (52.177.174.250), connection timed out

E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
Fetched 1,352 kB in 2min 0s (11.2 kB/s)
##[error]Process completed with exit code 100.
```

As an example: https://github.com/adafruit/circuitpython/actions/runs/68744396

This build has 123 checks, of which 10 failed because Travis CI couldn't install dependencies.  As a contributor, there isn't any way for me to re-run these failed tests without making changes and re-pushing, which risks seeing different failures appear.

jepler, 2020-04-02:
BTW, our CI is now Github Actions, not Travis.  I think the actual problem leading to the "unable to connect" / "could not connect" message may be some kind of capacity/reliability problem of azure.

Because of other problems with Github Actions infrastructure, even project maintainers get no love when they use the "re-run all jobs" button, because when you re-run a PR job it just errors at the checkout phase.  This non-working re-run button is a bug (an acknowledged bug, I think) in actions/checkout@v1.  I had attempted to convert us to actions@v2 but that ended up failing with a different kind of impossible-looking error (#2725, with just a few percent of builds failing with "fatal: 'origin' does not appear to be a git repository" while fetching tags in submodules). Even so, that would have only made the re-run all jobs button work, not fix the underlying reliability problem in azure/github actions.

https://github.com/python-pillow/Pillow/issues/4508
https://github.community/t5/GitHub-Actions/sudo-apt-get-install-fails-with-Could-not-connect-to-azure/td-p/52300
https://github.community/t5/GitHub-Actions/Install-dependencies-in-Ubuntu-flakes-several-times-a-wekk/td-p/51785
https://github.community/t5/GitHub-Actions/sudo-apt-install-fails-with-Unable-to-connect-to-azure-archive/td-p/32154

xobs, 2020-04-02:
Thanks, I'm not sure why I thought it was Travis.  My mistake.  I somehow thought it was backed by Travis.  I'm learning new things all the time.

The error in #2725 looks like it may be because you're only fetching tags and refs, which won't fetch dangling references that have no tags (for example, if a later commit rebased).  My recipe is usually `git fetch; git checkout [ref]`.

Us users don't have access to a "Re-run job" button.

If Azure is unreliable, is it possible to add a different repository to pull from?

sommersoft, 2020-04-04:
Just to keep this issue "in the loop", an issue has been posted to [`actions/virtual-environments`](https://github.com/actions/virtual-environments/issues/675) regarding the `azure.archive.ubuntu.com` repository failures.

Should we further edit this issue title to specify _which_ Actions failure this pertains to? Since we have a few we're working through...

dhalbert, 2020-04-04:
@sommersoft Maybe a few pointers to some failed jobs would be helpful to them?

sommersoft, 2020-04-04:
There are pointers to failed jobs in the GitHub Community forum posts. Is it bad form to keep plastering them in each place? 😄

dhalbert, 2020-04-04:
OK, then probably not necessary; they just need to fix it. :smiley:

tannewt, 2020-07-14:
I haven't seen this recently and suspect GitHub Actions as the issue. So, closing this.

closed 2020-07-14 by tannewt