adafruit/circuitpython #2746: Itsy Bitsy M0 Express Audio Noise when accelerometer connects

brettm2, 2020-04-02:
I've had noise issues with the Itsy Bitsy M0 Express and have boiled it done to attaching the MSA301 Accelerometer. I can attach neopixels, buttons, etc. but as soon as I attach the accelerometer I get random loud noise. I've also tried different speakers and audio boards with no change in the noise. The noise can occur right when I turn it on or it may wait 3-4 sec before starting.

Here is the simplified code:
```
# Import libraries
import board
from digitalio import DigitalInOut, Direction, Pull
import busio
import adafruit_msa301
import audioio
import audiobusio

# initialize audio
audio = audiobusio.I2SOut(board.D1, board.D0, board.D9)

# Accelerometer
i2c = busio.I2C(board.SCL, board.SDA)
msa = adafruit_msa301.MSA301(i2c)
msa.range = adafruit_msa301.Range.RANGE_16_G

#Initialize button
button1 = DigitalInOut(board.D10)
button1.direction = Direction.INPUT
button1.pull = Pull.UP

def play_wav(name, loop=False, sample_rate=2000):
    try:
        wave_file = open('sounds/' + name + '.wav', 'rb')
        wave = audioio.WaveFile(wave_file)
        audio.play(wave, loop=loop)
    except:
        return

while True:
    if not button1.value:
        while not button1.value:
            pass
        play_wav("hum",True)

    if audio.playing:
        x, y, z = msa.acceleration
```

Here is a wiring diagram:
[image]

I'm pretty new at adding sound to projects so I'm hoping it's something simple I'm not seeing. Any help figuring this out would be greatly appreciated.

tannewt, 2020-07-14:
@caternuson want to see if you can reproduce this?

caternuson, 2020-07-14:
@tannewt Yep, recreated. I went back to just the basic setup per the guide:
https://learn.adafruit.com/adafruit-max98357-i2s-class-d-mono-amp/circuitpython-wiring-test
and still see the issue. So may be something more fundamental and not related to MSA301.

[image]

Tested as follows. Libraries from 20200714 bundle release.

```python
Adafruit CircuitPython 5.3.1 on 2020-07-13; Adafruit ItsyBitsy M0 Express with samd21g18
>>> import audioio
>>> import board
>>> import audiobusio
>>> wave_file = open("StreetChicken.wav", "rb")
>>> wave = audioio.WaveFile(wave_file)
>>> audio = audiobusio.I2SOut(board.D1, board.D0, board.D9)
>>> audio.play(wave)
>>> audio.play(wave)
>>> audio.play(wave)
>>> audio.play(wave)
>>> 
```

The issue isn't there all the time. Some of the playbacks above were OK. Some (most) were distorted.

tannewt, 2020-07-14:
Thanks @caternuson! @jepler any ideas what could cause this?

daveythacher, 2021-04-12:
See page 33 of datasheet. Try with Pi.
https://cdn-learn.adafruit.com/assets/assets/000/032/642/original/adafruit_products_schem.png?1464034817

dhalbert, 2021-04-12:
@daveythacher Could you be more explicit? Which datasheet? Are you saying something about the filtering?

daveythacher, 2021-04-12:
The datasheet:
https://cdn-shop.adafruit.com/product-files/3006/MAX98357A-MAX98357B.pdf

It is possible due to wiring and ground noise for there to be a some disruption in sound quality. There are filters on the output per the datasheet for long speaker wires. There are some bank capacitors also per datasheet. The IC is designed to support low noise production and has measures on board to limit noise from making it to the output. However some of these are automatic and work in certain places better than in others.

The datasheet suggests using more capacitors near the IC, which is hard to do here. The datasheet suggests handling ground and power carefully. That is only done to some extend. Long wires, breadboards and limited decoupling capacitance can cause issues.

If my memory serves me this has happened to me on Pi. I blamed my wiring and setup. Generally speaking these are more resilient to noise, however they can still be susceptible. So first I would recommend seeing if you can reproduce the problem outside of circuitPython. My thinking is you will based on the diagram posted by @caternuson.

Some of these microcontrollers have fast switch currents of different frequencies which may leak. The wires could also pick this up. I am not an expert in this area. However my guess is that there was some kind of change caused by the accelerometer due to wiring or noise. Based on @caternuson setup the code was running in a fairly ideal condition for runtime, which means that the logic is not likely the issue.

It is worth reading over the datasheet were it talks about the filtering that is supported on board. It mentions cutoff frequencies and automatic mode switches. Which may also explain why certain tracks perform better.

The MCU, DAC, accelerometer, noise, etc. could all create an issue. I know for full up products you need to carefully design your circuits to suppress EMI. I do not believe this to be done on all boards for cost considerations. They are designed for evaluation purposes only.

Cleaning up the circuit may help. Remove breadboard, add decoupling capacitors, shorten the wires, route the wires around switching logic, etc. Again not my area, There is a lot of analog, signal and time domain stuff that can happen.

dhalbert, 2023-10-28:
Increasing the buffer size may help. We saw similar issues on RP2040 PropMaker.

ValleyKX, 2024-02-22:
Just a heads up: The board you have doesn't support I2S, and you're attempting to use I2S to run the speaker. I ran into this problem and figured I'd leave this here in case anyone else runs into this issue.