adafruit/circuitpython #2748: FrequencyIn returns 0 when fed 500 Hz

tannewt, 2020-04-07:
I'm feeding a Grand Central M4 500 Hz on D6 and it's returning 0 for the value.

```
Adafruit CircuitPython 5.1.0 on 2020-04-06; Adafruit Grand Central M4 Express with samd51p20
>>>
```

Test code.py:
```python
import frequencyio
import time
import board

frequency = frequencyio.FrequencyIn(board.D6)
while True:
    print(frequency.value)
    time.sleep(1)
```

I have a metro m4 generating the 500Hz signal and verified it is working with a Saleae.

tannewt, 2020-04-09:
I tried this with a 1 MHz signal and it also returned zero for me.

sommersoft, 2020-04-15:
Just tried this on 5.2.0, with a Feather M0 piping `PWMOut` into a Metro M4:
```python
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.2.0 on 2020-04-09; Adafruit Metro M4 Express with samd51j19
>>> import frequencyio
>>> import board
>>> import time
>>>
>>> freq = frequencyio.FrequencyIn(board.D0)
>>> while True:
...     print(freq.value)
...     time.sleep(1)
...
0         # Feather freq out: 500Hz
0
0
0
0
0
0
1087      # Feather freq out: 1KHz
1087
1087
1087
1087
1087
2000      # Feather freq out: 2KHz
2200
2000
2000
2100
2000
2000
2100
5333900   # Feather freq out: 5MHz
5332800
5333600
5333300
5333500
5333900
5333800
5333100
5333300
```

I'll try `lower_power` branch after getting it built, to test there.

@tannewt, how was your signal generated? Not doubting its correct as verified by the Saleae. Just wondering if the EVSYS isn't getting triggered... (not sure how it wouldn't be, but worth an ask).

sommersoft, 2020-04-16:
Worked for me built with current #2685 . 

```py
Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.1.0-117-ga6fb0beaa on 2020-04-15; Adafruit Metro M4 Express with samd51j19
>>> import frequencyio
>>> import time
>>> import board
>>> freq = frequencyio.FrequencyIn(board.D0)
>>> while True:
...     print(freq.value)
...     time.sleep(1)
...
0          # Feather freq out: 500Hz
0
0
0
1000       # Feather freq out: 1KHz
1000
1000
1000
1000
2100       # Feather freq out: 2KHz
2001
2001
2001
2001
999899     # Feather freq out: 1MHz
999800
999800
999899
1066600
999899
999899
5332900    # Feather freq out: 5MHz
5333000
5333100
5332400
5688300
```

sommersoft, 2020-04-16:
Found a thread to pull on... Looks like Grand Central `D6` doesn't have a TC, which _should_ be throwing an error.

https://github.com/adafruit/circuitpython/blob/master/ports/atmel-samd/boards/grandcentral_m4_express/pins.c#L35
```
    { MP_OBJ_NEW_QSTR(MP_QSTR_D6),  MP_ROM_PTR(&pin_PD20) },
```

https://github.com/adafruit/samd-peripherals/blob/master/samd/samd51/pins.c#L824
```
#ifdef PIN_PD20
PIN(PD20, EXTINT_CHANNEL(10), NO_ADC, NO_ADC,
    NO_TOUCH,
    SERCOM(1, 2),
    SERCOM(3, 2),
    NO_TIMER,
    TCC(1, 0),
    NO_TIMER);
#endif
```

sommersoft, 2020-04-16:
Nothing is jumping out at me in the codebase. I've walked the register settings against what the Grand Central should have from ASF4/samd-peripherals; can't see anything that would be set incorrectly.

I don't have a GC, so I can't verify the registers; would be my next step if I did have one.
DigiKey has them in stock, so I may just pick one up. Having that many GPIO could surely come in handy for something. 😄