adafruit/circuitpython #2755: GD25Q127C Working, But Not Recognized Adafruit_QSPI library ...

Casey10110, 2020-04-08:
Hi Guys,
I finally managed to get a 16MB (largest possible for SAMD51) flash device working on a Feather M4 Express-based board. I used this:

`#define GD25Q127C {\
    .total_size = (1 << 24), /* 16 MiB */ \
    .start_up_time_us = 5000, \
    .manufacturer_id = 0xc8, \
    .memory_type = 0x40, \
    .capacity = 0x18, \
    .max_clock_speed_mhz = 104, /* if we need 120 then we can turn on high performance mode */ \
    .quad_enable_bit_mask = 0x02, \
    .has_sector_protection = false, \
    .supports_fast_read = true, \
    .supports_qspi = true, \
    .supports_qspi_writes = true, \
    .write_status_register_split = true, \
    .single_status_byte = false, \
}`

I know it works, as I see the correct flash size and can upload files to it when I overwrite the uf2 with the circuitpython uf2. The problem is, when using the Adafruit QSPI library's flash_info example, it doesn't recognize it. Any ideas why? I added the same thing in "external_devices.h" in that library and added additional flash devices here in Adafruit_QSPI_Flash.cpp:

`static const external_flash_device possible_devices[] =
{
  GD25Q16C, GD25Q64C, 
  GD25Q127C,    // Main devices current Adafruit
  S25FL116K, S25FL216K,
  W25Q16FW, W25Q64JV_IQ, // Only a handful of production run

  MX25R6435F,            // Nordic PCA10056
};`

Any ideas what the issue could be?

tannewt, 2020-04-09:
This isn't a CircuitPython issue. Adafruit_QSPI is deprecated in favor of Adafruit_SPIFlash (https://github.com/adafruit/Adafruit_QSPI). So, please try in the new library and file an issue on Adafruit_SPIFlash if it doesn't work there.

closed 2020-04-09 by tannewt

Casey10110, 2020-04-09:
> Links:

Awesome, I added it in there and now it works. I didn't realize I wasn't 
using the QSPI library. Somebody might officially add this chip to 
circuitpython and SPI_Flash, as the other ones this large in size did 
not work for me at all. It's the largest you can use with a SAMD51:

#define GD25Q127C {\
     .total_size = (1 << 24), /* 16 MiB */ \
     .start_up_time_us = 5000, \
     .manufacturer_id = 0xc8, \
     .memory_type = 0x40, \
     .capacity = 0x18, \
     .max_clock_speed_mhz = 104, \
     .quad_enable_bit_mask = 0x02, \
     .has_sector_protection = false, \
     .supports_fast_read = true, \
     .supports_qspi = true, \
     .supports_qspi_writes = true, \
     .write_status_register_split = true, \
     .single_status_byte = false, \
}

Best,
Casey

hathach, 2020-04-09:
@Casey10110 I am maintainer of the Arduino SPIFlash lib, if you could submit an PR to add flash device this to the list. I will be happy to merge it there.