adafruit/circuitpython #2759: Add support for the seeed wio terminal

olivier-boesch, 2020-04-10:
Hello,

This new board embeds a atsamd51 so it could be possible to support it easily.

It also has a realtek 8720dn as a wireless coprocessor (not sure). maybe a library coded in python could enable wireless com.

What do you think about it?

I ordered one and will be happy to act at least as a tester.

ladyada, 2020-04-10:
yes! please ask seeed to submit a PR! :)

olivier-boesch, 2020-04-10:
This is done...

ladyada, 2020-04-10:
great, please follow up with them later :)

olivier-boesch, 2020-04-11:
The answer speaks by itself :

"Forget circuitpython, our ArduPy will make you feel the magic of programming again."
But I don't agree.

PTS93, 2020-04-13:
wow.. is that a public github issue you could link to or was that via email?

olivier-boesch, 2020-04-13:
That's a public post on a seeed forum. see by yourself:

https://forum.seeedstudio.com/t/new-releasing-of-wio-terminal-battery-chassis-and-look-forward-to-your-ideas/251049/7

tannewt, 2020-04-13:
@olivier-boesch We'd be happy to help you add support to CircuitPython. Just let us know when we can!

tannewt, 2020-08-18:
@ansonhe97 added the board. Wifi isn't supported though.

closed 2020-08-18 by tannewt