adafruit/circuitpython #2760: fail to build fomu, undefined reference

gojimmypi, 2020-04-11:
When building Circuit Python in WSL for the FOMU, I am seeing a bunch of `undefined reference` errors. I'm following the basic instructions from [here](https://learn.adafruit.com/building-circuitpython/build-circuitpython):

```
git clone https://github.com/adafruit/circuitpython.git
cd circuitpython
git submodule sync
git submodule update --init
make -C mpy-cross
```
and when I do a make:
```
cd /mnt/c/workspace/circuitpython/ports/litex
make BOARD=fomu
```
I see these errors:
```
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: section `.bss' type changed to PROGBITS
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: build-fomu/py/runtime.o: in function `mp_binary_op':
/mnt/c/workspace/circuitpython/ports/litex/../../py/runtime.c:425: undefined reference to `__mulsi3'
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /mnt/c/workspace/circuitpython/ports/litex/../../py/runtime.c:442: undefined reference to `__floatsisf'
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /mnt/c/workspace/circuitpython/ports/litex/../../py/runtime.c:442: undefined reference to `__floatsisf'
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /mnt/c/workspace/circuitpython/ports/litex/../../py/runtime.c:458: undefined reference to `__floatsisf'
... [snip many similar errors for __mulsf3, __eqsf2, __subsf3, __addsf3, etc]
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: build-fomu/firmware.elf(.text): relocation ".L15+0x0 (type R_RISCV_RVC_JUMP)" goes out of range
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/libgcc.a(divsf3.o): file class ELFCLASS64 incompatible with ELFCLASS32
/opt/riscv/lib/gcc/riscv64-unknown-elf/9.2.0/../../../../riscv64-unknown-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
Makefile:177: recipe for target 'build-fomu/firmware.elf' failed
make: *** [build-fomu/firmware.elf] Error 1
```
This is my RISC-V compiler:
```
$  riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (GCC) 9.2.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
Any suggestions what I might be doing wrong? Perhaps a missing `apt-get install`? Thanks.

tannewt, 2020-04-11:
These errors don't look familiar to me. Are you using the [Fomu toolchain](https://github.com/im-tomu/fomu-toolchain/releases) or your own RISC-V toolchain?

gojimmypi, 2020-04-11:
Although I had the `riscv32i` for the tinyFPGA and ULX3S soft SoC for LiteX, the Circuit Python here for the fomu wanted the 64 bit `riscv64-unknown-elf-gcc`, so I built it from [riscv/riscv-gnu-toolchain](https://github.com/riscv/riscv-gnu-toolchain/blob/master/README.md#installation-newlib). 

I assume I should be using the fomu version, instead?

gojimmypi, 2020-04-11:
The answer is YES, the [Fomu toolchain](https://github.com/im-tomu/fomu-toolchain/releases) works great! Thank you @tannewt for that tip!! I used the [linux_x86_64-v1.5.6.tar.gz](https://github.com/im-tomu/fomu-toolchain/releases/download/v1.5.6/fomu-toolchain-linux_x86_64-v1.5.6.tar.gz).

```
gojimmypi:/mnt/c/workspace/circuitpython/ports/litex
$ make BOARD=fomu
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
/opt/fomu-toolchain-linux_x86_64-v1.5.6/bin/../lib/gcc/riscv64-unknown-elf/8.3.0/../../../../riscv64-unknown-elf/bin/ld: warning: section `.bss' type changed to PROGBITS

739752 bytes free in flash firmware space out of 1048576 bytes (1024.0kB).
116920 bytes free in ram for stack and heap out of 131072 bytes (128.0kB).

Create build-fomu/firmware.dfu
python3 ../../tools/dfu.py -b build-fomu/firmware.bin -D 0x1209:0x5bf0 "build-fomu/firmware.dfu"
```
Although that `riscv64-unknown-elf-gcc` is a bit older version:

```
$ riscv64-unknown-elf-gcc --version
riscv64-unknown-elf-gcc (SiFive GCC 8.3.0-2019.08.0) 8.3.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
I'm going to leave this issue open, as why would my build of the compiler fail? Problem specific to version 9.2.0? Probably not. I suspect there are dependency include files in the fomu gz file that should probably be documented on the respective [README.md](https://github.com/im-tomu/fomu-toolchain/blob/master/README.md).

In any case: Woohoo! I have Circuit Python building for the FOMU!! Next step: ULX3S.

gregdavill, 2020-04-14:
Looks like you were using a riscv64 compiler that didn't have either the 32bit ABI enabled, or multilib. If you were building gcc from scratch it can be a tricky to get all the build options correct.

Do you have that compiler still installed? Might be worth checking some options? For reference this is my gcc
```
riscv64-unknown-elf-gcc --help=target
The following options are target specific:
  -mabi=                      Specify integer and floating-point calling convention.
  -march=                     Generate code for given RISC-V ISA (e.g. RV64IM).  ISA strings must be lower-case.
  -mbranch-cost=N             Set the cost of branches to roughly N instructions.
  -mcmodel=                   Specify the code model.
  -mdiv                       Use hardware instructions for integer division.
  -mexplicit-relocs           Use %reloc() operators, rather than assembly macros, to load addresses.
  -mfdiv                      Use hardware floating-point divide and square root instructions.
  -mplt                       When generating -fpic code, allow the use of PLTs. Ignored for fno-pic.
  -mpreferred-stack-boundary= Attempt to keep stack aligned to this power of 2.
  -mrelax                     Take advantage of linker relaxations to reduce the number of instructions required to materialize symbol addresses.
  -msave-restore              Use smaller but slower prologue and epilogue code.
  -msmall-data-limit=N        Put global and static data smaller than <number> bytes into a special section (on some targets).
  -mstrict-align              Do not generate unaligned memory accesses.
  -mtune=PROCESSOR            Optimize the output for PROCESSOR.

  Supported ABIs (for use with the -mabi= option):
    ilp32 ilp32d ilp32f lp64 lp64d lp64f

  Known code models (for use with the -mcmodel= option):
    medany medlow
```

```
riscv64-unknown-elf-gcc -v 
Using built-in specs.
COLLECT_GCC=riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv64/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/bin/../libexec/gcc/riscv64-unknown-elf/8.1.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /scratch/carsteng/rbt-lto-python/obj/x86_64-linux-ubuntu14/build/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --host=x86_64-linux-gnu --prefix=/scratch/carsteng/rbt-lto-python/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14 --disable-shared --disable-threads --enable-languages=c,c++ --without-system-zlib --enable-tls --with-newlib --with-sysroot=/scratch/carsteng/rbt-lto-python/obj/x86_64-linux-ubuntu14/install/riscv64-unknown-elf-gcc-8.1.0-2019.01.0-x86_64-linux-ubuntu14/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --src=.././riscv-gcc --enable-checking=yes --enable-multilib --with-abi=lp64d --with-arch=rv64imafdc 'CFLAGS_FOR_TARGET=-Os  -mcmodel=medany' 'CXXFLAGS_FOR_TARGET=-Os  -mcmodel=medany'
Thread model: single
gcc version 8.1.0 (GCC)
```

gojimmypi, 2020-04-14:
@gregdavill yes, I had built the RISC-V from scratch, using the [basic setup](https://github.com/riscv/riscv-gnu-toolchain/blob/master/README.md#installation-newlib) (no special `--with` parameters) see also [my blog on this topic](https://gojimmypi.blogspot.com/2020/04/circuit-python-on-fomu-in-wsl-plus.html)

I still have it installed, although the @im-tomu pre-compiled binaries are listed first in my path. Here are details on that scratch-built one:
```
$ /opt/riscv/bin/riscv64-unknown-elf-gcc --help=target
The following options are target specific:
  -mabi=                      Specify integer and floating-point calling convention.
  -malign-data=               Use the given data alignment.
  -march=                     Generate code for given RISC-V ISA (e.g. RV64IM).  ISA strings must be lower-case.
  -mbranch-cost=N             Set the cost of branches to roughly N instructions.
  -mcmodel=                   Specify the code model.
  -mdiv                       Use hardware instructions for integer division.
  -mexplicit-relocs           Use %reloc() operators, rather than assembly macros, to load addresses.
  -mfdiv                      Use hardware floating-point divide and square root instructions.
  -mplt                       When generating -fpic code, allow the use of PLTs. Ignored for fno-pic.
  -mpreferred-stack-boundary= Attempt to keep stack aligned to this power of 2.
  -mrelax                     Take advantage of linker relaxations to reduce the number of instructions required to
                              materialize symbol addresses.
  -mriscv-attribute           Emit RISC-V ELF attribute.
  -msave-restore              Use smaller but slower prologue and epilogue code.
  -msmall-data-limit=N        Put global and static data smaller than <number> bytes into a special section (on some
                              targets).
  -mstrict-align              Do not generate unaligned memory accesses.
  -mtune=PROCESSOR            Optimize the output for PROCESSOR.

  Supported ABIs (for use with the -mabi= option):
    ilp32 ilp32d ilp32e ilp32f lp64 lp64d lp64f

  Known code models (for use with the -mcmodel= option):
    medany medlow

  Known data alignment choices (for use with the -malign-data= option):
    natural xlen

```

```
$ /opt/riscv/bin/riscv64-unknown-elf-gcc -v
Using built-in specs.
COLLECT_GCC=/opt/riscv/bin/riscv64-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv64-unknown-elf/9.2.0/lto-wrapper
Target: riscv64-unknown-elf
Configured with: /mnt/c/workspace/riscv-gnu-toolchain/riscv-gcc/configure --target=riscv64-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv64-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=.././riscv-gcc --disable-multilib --with-abi=lp64d --with-arch=rv64imafdc --with-tune=rocket 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
gcc version 9.2.0 (GCC)
```

gregdavill, 2020-04-14:
I see your build has `--disable-multilib` in the configured with section. This means there is no newlib built for rv32i, which CircuitPython.fomu depends on.

I think if you were to try re-compiling gcc with `--enable-multilib` argument in the configure command it should work.

tannewt, 2021-04-08:
Is there anything to be done here?

dhalbert, 2021-06-15:
Closing this now.

closed 2021-06-15 by dhalbert