adafruit/circuitpython #2762: Recursive clone of circuitpython fails due to bad submodule path

larsks, 2020-04-11:
The circuitpython repository includes the following submodule:

```
[submodule "extmod/ulab"]
        path = extmod/ulab
        url = https://github.com/v923z/micropython-ulab/
```

The trailing slash on the repository URL causes a failure when trying to clone it:

```
$ git clone https://github.com/v923z/micropython-ulab/
Cloning into 'micropython-ulab'...
ERROR: Repository not found.
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
```

This results in a failure when trying to clone circuitpython using `git clone --recurse-submodules` or when running `git submodule update`.

closed 2020-04-11 by dhalbert

jepler, 2020-04-11:
I'm curious what git versions were affected by this.  Locally I have git version 2.20.1 and did not observe the problem:
```
$ git clone https://github.com/v923z/micropython-ulab/
Cloning into 'micropython-ulab'...

remote: Enumerating objects: 103, done....
```

larsks, 2020-04-11:
@jepler Turns out it's not a git version issue, but it's a side effect of my `git` configuration, which has this:

```
[url "ssh://git@github.com/"]
        insteadof = https://github.com/
```

That transforms `http://` access into `ssh://` access, and while `git@github.com:v923z/micropython-ulab` is a valid url, `git@github.com:v923z/micropython-ulab/`.

jepler, 2020-04-11:
Aha, that would do it.  I'm glad you were able to diagnose and fix the problem, thanks much!

PR #2763 (2020-04-11): Fix bad submodule path
The repository URL for extmod/ulab had a trailing slash, causing
errors when trying to clone it.

Closes #2762
commit abf79de7 (2020-04-11): Fix bad submodule path

The repository URL for extmod/ulab had a trailing slash, causing
errors when trying to clone it.

Closes #2762