adafruit/circuitpython #2769: cpu_get_regs_and_sp(regs) only uses sp

xobs, 2020-04-14:
The `cpu_get_regs_and_sp(regs)` function appears to be used in two places, and the `regs` argument is completely unused.  Additionally, the `regs` argument is hardcoded several places to be `10` elements long, which doesn't work for platforms such as RISC-V which has 31 general registers.

Unless this function has other uses I'm not aware of, we can increase portability and remove unused code by renaming this function `cpu_get_sp()` and removing the argument.

tannewt, 2020-04-14:
I'm not sure what the right answer is. I think for correctness we need to collect pointers in the registers so that we don't gc something we're still using. However, that may lead to use hanging onto stuff we no longer need.

tannewt, 2026-10-13:
We no longer support the litex port.

closed 2026-10-13 by tannewt