adafruit/circuitpython #2773: Audio problems with stage game library

GeyerJoachim, 2020-04-15:
CircuitPython 5 seems to have broken the playing of audio files within the stage/ugame-library (https://learn.adafruit.com/circuitpython-stage-game-library/overview). WAV-Files sound strange and too slow on a pygamer or pybadger (for example with this game: https://github.com/python-ugame/vacuum-invaders). There was no such problem with CircuitPython 4.
Thank you,
Joachim Geyer

deshipu, 2020-04-15:
I asked to report it here, because Stage simply passes audio to the ``audioio`` module, and doesn't really do anything by itself, so it seems like an ``audioio`` problem. However, I didn't have time to verify it myself.

tannewt, 2020-04-15:
@jepler please take a look next time you have cycles.

deshipu, 2020-04-16:
I'm sorry, it seems that is an issue with Stage after all. Looks like the timings have changed, and now the 64-byte buffer that I used for playing the sounds is not enough. Doubling it to 128 bytes fixes the sound. I will make a pull request.

jepler, 2020-04-16:
I'm glad you figured it out.  We (deliberately) decreased the frequency with which background tasks could run, in 5.0, to be no more often than 1ms.  Of course, other things like GC pauses can make the interval >>1ms anyway, unless your code very carefully has no heap allocations.

commit f7e89879 (2020-04-20): Bump circuitpython-stage to 1.0.11 to fix audio problems

Fix #2773
closed 2020-04-20 by tannewt