adafruit/circuitpython #2781: Add swdio

tannewt, 2020-04-17:
We need a low level `swdio` module that does the lowest level of SWD using bitbanging. That way we'll be able to use CircuitPython to recover another board that had it's bootloader busted.

SWD overview is here: https://github.com/MarkDing/swd_programing_sram

tannewt, 2020-04-17:
This is also useful but we don't need the SPI speed yet: https://medium.com/@ly.lee/openocd-on-raspberry-pi-better-with-swd-on-spi-7dea9caeb590

We'll only use it to re-flash the uf2 bootloader.

k0d, 2020-04-20:
I've been writing a CMSIS-DAP debugger app on top of Zephyr...so I have quite a bit of knowledge of this. How do you think it should work? A python lib/script that they just copy over MSC to the 'host' device and that connects to the 'target' device to program it?

ladyada, 2020-04-20:
ya it would be heavily based on/perhaps even wrapper of https://github.com/adafruit/Adafruit_DAP which does buffers and fuses. then you can read/write from disk if ya like

tannewt, 2020-04-20:
@k0d I think `swdio` would just be SWD primitives that are time sensitive. On top of that we can write Python logic to change behavior based on the chip and flash setup.

tannewt, 2023-07-07:
Bitbanged here: https://github.com/adafruit/Adafruit_CircuitPython_Debug_Probe

Higher level flashing protocols (like Adafruit_DAP) are here: https://github.com/adafruit/Adafruit_CircuitPython_MCU_Flasher