adafruit/circuitpython #2783: Drawing displayio elements with loaded BDFs very slow

k3wals, 2020-04-18:
Testing scope: 5.0.0 and 5.1.0 releases on Feather NRF52840 Express

I'm using a [240x240 IPS display](https://www.adafruit.com/product/4520), and my display layout consists of 

- 4 groups
- 23 labels (total)
- four lines (total)
- one Bitmap

I'm loading three fonts like this:

```
font_arial = bitmap_font.load_font("/fonts/Arial.bdf")
font_arial_16 = bitmap_font.load_font("/fonts/Arial-16.bdf")
font_helvetica_bold_16 = bitmap_font.load_font("/fonts/Helvetica-Bold-16.bdf")
# font_arial.load_glyphs(b'abcdefghjiklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ1234567890- -.,:()')
# font_arial_16.load_glyphs(b'abcdefghjiklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ1234567890- -.,:()')
# font_helvetica_bold_16.load_glyphs(b'abcdefghjiklmnopqrstuvwxyzABCDEFGHIJKLMNOPQRSTUVWXYZ1234567890- -.,:()')
```
The load_glyphs were an experiment to try to see an improvement but there wasn't. Now they eat up too much memory due to application growth.

The font loading itself takes less than a second, which is great. The rendering of the labels takes **21 seconds!** Turning off refresh during this time doesn't help either.

```
main.py output:
starting font load at  4/18/2020 0:37:3
ending font load at  4/18/2020 0:37:3
starting splashscreen setup load at  4/18/2020 0:37:3
ending splashscreen setup load at  4/18/2020 0:37:6
starting main screen setups load at  4/18/2020 0:37:7
ending main screen setups load at  4/18/2020 0:37:14
starting events screen setups load at  4/18/2020 0:37:14
ending events screen setups load at  4/18/2020 0:37:24
starting event screen setups load at  4/18/2020 0:37:24
ending event screen setups load at  4/18/2020 0:37:24
The current time is: 4/18/2020 00:37:24

Result: 21 seconds
```
If I switch to terminalio.FONT temporarily, the loading is nearly instant. I thought _bitmap_font.load_font()_ loaded glyphs to memory, so what could be the bottleneck?

[cp_labels_slow_code_and_fonts.zip](https://github.com/adafruit/circuitpython/files/4496164/cp_labels_slow_code_and_fonts.zip)

k3wals, 2020-04-18:
As an experiment, I hacked up the code to allow baking in more than one font to create custom _terminalio.FONT_ and _terminalio.FONT2_

I baked in Arial.bdf and Helvetica-Bold-16.bdf. Arial-16 only had two labels so I changed them to FONT2.

The resulting load time is **4 seconds**, of which 3 is the bitmap logo.

[patch_circuitpython5.1.0_2fonts_hack.txt](https://github.com/adafruit/circuitpython/files/4496185/patch_circuitpython5.1.0_2fonts_hack.txt)

k3wals, 2020-04-18:
As an experiment, I hacked up the code to allow baking in more than one font to create custom _terminalio.FONT_ and _terminalio.FONT2_

I baked in Arial.bdf and Helvetica-Bold-16.bdf. Arial-16 only had two labels so I changed them to FONT2.

The resulting load time is **4 seconds**, of which 3 is the bitmap logo.

https://github.com/k3wals/circuitpython/commit/3d3e6b16f0ed024003b16280a06c7aaa3d1a1193

[patch_circuitpython5.1.0_2fonts_hack.txt](https://github.com/adafruit/circuitpython/files/4496185/patch_circuitpython5.1.0_2fonts_hack.txt)

tannewt, 2020-04-20:
Hi hi, load_font does not load the glyphs into memory because it would load a bunch it doesn't need. load_glyphs is a way to do one linear search of the font file for all characters of interest.

Passing the font to a label will cause each new glyph to be loaded independently which is really slow because it is a linear search through a text file for each glyph. I recommend doing load_glyphs to speed things up.

Another way to speed things up would be using a file format that we could jump through and that wasn't string parsing.

k3wals, 2020-04-21:
My prior "experiment" utilizing _load_glyphs_ must have been incomplete because I see a reduction in speed. 

2x fonts with just load_fonts: 17 seconds
2x fonts with load_fonts and load_glyphs: 5 seconds
2x fonts baked into UF2: **1 second**

What's special about the fonts baked into the firmware image? Aren't those also copied from flash to ram at some point?  Why does the display element drawing go from 5 seconds to 1 second?

There isn't really an issue here since I'm happy with the "baked in" method, as I happen to call it. I suppose there's room for improvement with the adafruit_bitmap_font library.

Feel free to close, unless you want to keep open if there's room for improvement.

tannewt, 2020-04-21:
Baked in fonts aren't loaded into memory. They are objects in flash that are readily available.

I assume the 5 seconds is from load_glyphs which is doing a lot of text parsing. Our goal was using a wider variety of fonts, not to do it quickly. :-)

I'll close this issue because I think it's a library issue and not core CircuitPython. Thanks for the investigation!

closed 2020-04-21 by tannewt

chrishp, 2022-06-08:
Seems like the library is part of CircuitPython, this draw speed is unusable.  Imagine how often you need to write text and change its' value.

Could we have an official solution?

jepler, 2022-06-08:
Hi! For help with writing CircuitPython code, our Discord or the forums are better places to get assistance from the community.

Off the cuff, the main recommendation I'd make is to use "pcf" (binary format) fonts instead of "bdf". The time to 'load glyphs' is much less than with pcf fonts, though it's still not instant. In any case, we detail how to do this in a learn guide: https://learn.adafruit.com/custom-fonts-for-pyportal-circuitpython-display/convert-to-pcf

Second, depending on the application, a 'bitmap label' may perform better, but it varies. Again, we have a Learn guide about it: https://learn.adafruit.com/circuitpython-display_text-library/types-of-labels