adafruit/circuitpython #2790: Always build with -ggdb3

xobs, 2020-04-21:
Right now, debug symbols are only turned on when compiling with `DEBUG=1`.  I ran a test where I built the image twice: Once with `-ggdb3` and once with ` ` (i.e. adding a space to ensure both were built from dirty git repos).

The resulting file was 8 bytes larger over 228384 bytes, and much of the resulting binary was identical.  I suspect there may be other conflating factors including the order in which symbols are generated.

Is there a reason circuitpython isn't normally built with `-ggdb3`?

tannewt, 2020-04-21:
No, I think @jepler has wanted this too.

jepler, 2020-04-21:
I do.  Also, the step of always running 'strip' after linking circuitpython needs to be removed.

dhalbert, 2020-05-17:
I'm confused by this. a gdb build normally has many symbol strings in it, which makes it a lot larger.  Are we talking about the .elf only? We of course don't want these strings in the .bin/.uf2.

xobs, 2020-05-17:
A gdb build normally has many symbol strings, however it's missing various `.debug_*` sections.  These sections can be used to go from `$pc` locations back to function locations.  Without these sections you can kind of get a sense of where you are in the program, but if you're in a function that got inlined then the debugger has no idea where you are.  Additional information about symbol sizes may also be lost, and you won't get symbols such as preprocessor #define values.

These sections don't get copied to the `.bin` or `.uf2` file, they are only present in the `.elf` file.

Note that `-ggdb3` got accidentally included on the NRF port as part of 1d0781bf12e7b792842ea63cb986872ce61160f1.

dhalbert, 2021-01-02:
`-ggdb3` is now set explicitly  in `nrf`, `stm, `litex,` `esp32s2`, and `atmel-samd` Makefiles.

closed 2021-01-02 by dhalbert