adafruit/circuitpython #2792: Hard Crash - Feather m4 Express build on WSL1

wildestpixel, 2020-04-21:
Followed the guide at [https://learn.adafruit.com/building-circuitpython/build-circuitpython](https://learn.adafruit.com/building-circuitpython/build-circuitpython) to the letter and get what seems like a a good build output : 

```
pixel@RYZEN3500U:~/circuitpython/ports/atmel-samd$ make BOARD=feather_m4_express
Use make V=1, make V=2 or set BUILD_VERBOSE similarly in your environment to increase build verbosity.
QSTR updated

137512 bytes free in flash firmware space out of 499712 bytes (488.0kB).
164372 bytes free in ram for stack and heap out of 196608 bytes (192.0kB).

Converting to uf2, output size: 724480, start address: 0x4000
Wrote 724480 bytes to build-feather_m4_express/firmware.uf2

```
But when copied over to board it goes into a hard crash with the following serial output : 

```
You are in safe mode: something unanticipated happened.
CircuitPython core code crashed hard. Whoops!
Crash into the HardFault_Handler.
Press any key to enter the REPL. Use CTRL-D to reload.
```

Would appreciate any help

dhalbert, 2020-04-21:
Could you attach the .uf2 here? You may need to zip it up first.
What version of gcc are you using?
What does `git status say`? (I want to know which commit you're compiling.)

wildestpixel, 2020-04-21:
```
pixel@RYZEN3500U:~/circuitpython$ gcc --version
gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

```
pixel@RYZEN3500U:~/circuitpython$ git status
On branch master
Your branch is up to date with 'origin/master'.

nothing to commit, working tree clean
```

[firmware.zip](https://github.com/adafruit/circuitpython/files/4511737/firmware.zip)

dhalbert, 2020-04-21:
what does `arm-none-eabi-gcc --version` show?

wildestpixel, 2020-04-21:
```
pixel@RYZEN3500U:~/circuitpython$ arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

closed 2020-04-21 by tannewt

wildestpixel, 2020-04-21:
Confirmed building okay now. Many thanks for such a fantastic response.

PR #2797 (2020-04-21): displayio: Fix "bus type" problem introduced at 8cba145c9029
When allocate_display_bus_or_raise was factored out, the assignment of the bus's Python type was lost.  Restore it.

This would have affected displays of any type other than RGBMatrix, when they were created dynamically.  Boards with displays configured in flash were unaffected.

Testing performed: ran the code from https://forums.adafruit.com/viewtopic.php?f=60&t=164561 on a Metro M4 Express (no display attached)

Closes: #2792
commit 507e17fb (2020-04-21): displayio: Fix "bus type" problem introduced at 8cba145c9029

When allocate_display_bus_or_raise was factored out, the assignment
of the bus's Python type was lost.  Restore it.

This would have affected displays of any type other than RGBMatrix, when
they were created dynamically.  Boards with displays configured in flash
were unaffected.

Closes: #2792