adafruit/circuitpython #2793: Create makefile targets for flash/debug

k0d, 2020-04-21:
To enable easier debugging whilst developing, it could be good to have some kind of Makefile targets to flash the firmware and setup debugging etc. For now I'm using the following and setting the OPENOCD_FLAGS in the mpconfigboard.mk file. 

```
flash: $(BUILD)/firmware.elf
	@openocd $(OPENOCD_FLAGS) -c "program $< reset" -c "shutdown"

openocd:
	@openocd $(OPENOCD_FLAGS)

debug:
	@$(CROSS_COMPILE)gdb -ex "target extended-remote :3333" $(BUILD)/firmware.elf
```

tannewt, 2020-04-21:
I'd like to make sure this is namespaced to the tool such as `openocd_debug` because I use JLink instead of openocd.

hierophect, 2020-04-29:
I recently added a dfu-flash option to STM32 in #2819. Other ports that support it include NRF with a dfu-flash and the cxd56 which has custom flashing with their ".spk" file. 

This seems like something where there's going to be some natural port-level implementation differences. STM32 has the built in dfu bootloader that is always accessible and thus makes the most sense as the default `flash` command, compared to atmel and NRF which have an easier time fitting in a UF2 bootloader since it doesn't cramp their filesystem. So I don't know if standardizing across all ports has strict benefits.

k0d, 2020-04-29:
Thing is with the STM32 dfu bootloader is you need to bring high the boot0, but that's not so fun on the boards with stlink built in.

I think we should add a default target 'flash', which makes the most sense, but if there are more options, we can use 'flash-openocd', or 'flash-stlink' etc.

Personally, I'd like to copy the way Zephyr does it...it seems to work very well.

hierophect, 2020-04-29:
The question is which one is the default. If I recall correctly, Zephyr has one `west flash` and a list, defaulting to the top of the list unless you pass another option as a parameter `--jlink`, correct? I'm not sure how I would do that with a makefile.

k0d, 2020-04-29:
That sounds right about how Zephyr works, I think we could have a parameter, and also let it read from an ENV variable. I'd like to set CIRCUITPYTHON_DEFAULT_DEBUGGER=openocd for example.