adafruit/circuitpython #2795: Add ability to sleep until an alarm

tannewt, 2020-04-21:
#2685 only adds light sleep during `time.sleep`. We should add a new mechanic for sleeping until an alarm.

An alarm can be anything a microcontroller supports during sleep such as RTC alarm, pin change interrupt and comparator. The alarms should be in separate modules so that the microcontrollers can support different ones.

Then, one `sleep_until_alarm()` function can take any number of alarms. It will return the object that woke it up.

tannewt, 2020-09-17:
Here is an example:

```python
import supervisor
import pinalarm
import timealarm

pin0_alarm = pinalarm.PinLevelAlarm(board.IO0, True, pull=digitalio.Pull.UP)
time_alarm = timealarm.DurationAlarm(1.0)

while True:
    # start with a light sleep until an alarm triggers.
    alarm = supervisor.sleep_until_alarm(pin0_alarm, time_alarm)
    # do something based on what alarmed
    if alarm == pin0_alarm:
        print("pin change")
    elif alarm == time_alarm:
        print("time out")
```

dhalbert, 2020-11-18:
Subsumed by #2795. Closing and will copy over the example.

closed 2020-11-18 by dhalbert