adafruit/circuitpython #2798: Request .bin builds for uChip rather than .uf2?

mew-cx, 2020-04-21:
(not sure if this is the best venue?)

I'm trying to get CIrcuitPython running on the wonderfully tiny [Itaca uChip](https://circuitpython.org/board/uchip/) boards.  Those boards come with an Arduino Bossa-compatible bootloader.  I don't see a [UF2 bootloader](https://github.com/adafruit/uf2-samdx1/releases) availalble for uChip, but I'm fine with keeping the existing bootloader.

My understanding is installing CircuitPython with Bossa requires .bin files.  But only .uf2 files are available for uChip at the [S3 archive](https://adafruit-circuit-python.s3.amazonaws.com/index.html?prefix=bin/uchip/en_US/).

Unless I'm missing a another approach, may I suggest .bin files be created for uChip?  Or is there a tool for extracting the .bin from a .uf2?  Or am I off in the weeds?

Thanks
-- mew

tannewt, 2020-04-22:
Hi @mew-cx. I believe we build the bins. We just don't upload them. You can build it yourself with these instructions: https://learn.adafruit.com/building-circuitpython/

We can start uploading it once an exception is added here: https://github.com/adafruit/circuitpython/blob/master/tools/build_board_info.py#L42

mew-cx, 2020-04-23:
Thanks for the pointer to the build exceptions, @tannewt !   I'll submit a PR probably this weekend.

PR #2804 (2020-04-23): tools: upload .bin and .uf2 for Itaca uChip boards
uChip boards are shipped with an Arduino bootloader.  The .bin files can be installed without first installing a .uf2 bootloader.

Resolves #2798
closed 2020-04-23 by tannewt