adafruit/circuitpython #2812: Missing `time.time()` in Circuitpython 5.2.0

walchko, 2020-04-26:
Hello, according to the documentation [here](https://circuitpython.readthedocs.io/en/5.0.x/shared-bindings/time/__init__.html#time.time), there is supposed to be a `time.time()` function like normal python. However, when I try to use this function it doesn't exist. It is probably just a documentation oversight. My output from the error and REPL is shown below:

```
code.py output:
Traceback (most recent call last):
  File "code.py", line 37, in <module>
  File "code.py", line 17, in __init__
AttributeError: 'module' object has no attribute 'time'

Press any key to enter the REPL. Use CTRL-D to reload.
Adafruit CircuitPython 5.2.0 on 2020-04-09; Adafruit Trinket M0 with samd21e18
>>> import time
>>> time.time()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'module' object has no attribute 'time'
>>> dir(time)
['__name__', 'monotonic', 'sleep', 'struct_time']
>>> 
```

I am using a Trinket M0 and circuitpython 5.2.0.

sommersoft, 2020-04-26:
This is mostly a documentation issue, as it doesn't address `time.time()`'s need for [long integer support](https://github.com/adafruit/circuitpython/blob/652a5de2425849fe3a23442bbaa36445cd55ec3e/shared-bindings/time/__init__.c#L306).

Boards with small capacities/internal filesystems like the Trinket M0 and Gemma M0, have long integer support turned off to save space.

dhalbert, 2020-04-26:
I think it should be raising a `NotImplementedError` on non-longint builds.

DavePutz, 2020-05-05:
@dhalbert  - would you envision something like this?

>>> import time
>>> dir(time)
['__class__', '__name__', 'localtime', 'mktime', 'monotonic', 'monotonic_ns', 'sleep', 'struct_time', 'time']
>>> time.time()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No long integer support
>>> time.mktime('12:34')
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No long integer support
>>> time.localtime("12:34")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No long integer support
>>> time.monotonic_ns()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NotImplementedError: No long integer support

dhalbert, 2020-05-05:
Ideally, yes. The most space-saving way of doing this would be to have the implementations in `time_module_globals_table[]` all point to the same function that throws `NotImplementedError`, hwne there are no longints. The compiler might even do that merging automatically through optimization if they were broken out into separate but identical functions.

dhalbert, 2020-05-05:
Re your push not working; try merging from upstream, then do a make translate, then push. Or if it's something else, could you show the error message here?

theacodes, 2020-05-15:
This looks like it was resolved by #2871, so I'm going to close this. If that's not correct, feel free to reopen.

closed 2020-05-15 by theacodes