adafruit/circuitpython #2813: `microcontroller.nvm` missing on Trinket M0

walchko, 2020-04-26:
The documentation [here](https://circuitpython.readthedocs.io/en/2.x/shared-bindings/microcontroller/__init__.html#microcontroller.nvm) says `microcontroller.nvm` should return either `nvm.ByteArray` when available or `None`. I am was looking at my Trinket M0 (Circuitpython 5.2.0) and it gave me `AttributeError: 'Processor' object has no attribute 'nvm'` ... is this documentation error or an issue with the Trinket M0 build?

```
>>> import microcontroller as mcu
>>> mcu.cpu
<Processor>
>>> help(mcu.cpu)
object <Processor> is of type Processor
  frequency -- <property>
  temperature -- <property>
  uid -- <property>
  voltage -- <property>
>>> mcu.cpu.uid
bytearray(b'\xec=-\xf40TLPJ1.72\x07\t\xff')
>>> mcu.cpu.voltage
3.31917
>>> mcu.cpu.temperature
20.8005
>>> mcu.cpu.frequency
48000000
>>> mcu.cpu.nvm
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Processor' object has no attribute 'nvm'
```

rlourette, 2020-04-26:
From your comment, I would think it was mcu.nvm instead of mcu.cpu.nvm.

Stay well.

Rich

walchko, 2020-04-26:
Arghhh ... yes, you are right, I got confused

closed 2020-04-26 by walchko