adafruit/circuitpython #2815: Advice on submitting PR for linear encoders

LearnWeaver, 2020-04-27:
Hi all,
I'm a huge fan of this project. I was looking for some advice before I submit a PR.
I have a project that uses Optical Switches for determining wheel speed (and distance). I have put together a module that handles the counting via interrupts, very similar to rotaryio, however where rotaryio looks for a quadrature encoder, this module purely looks for single pulses.
My question is, how should I prepare my code for the PR. Given this linear encoder is similar to rotaryio, should I package it as a class LinearEncoder (to complement IncrementalEncoder). Or should this be put into a module like 'lineario'?

Sorry if I've not followed the process properly. First time contributing to a large open project like this.

tannewt, 2020-04-27:
@LearnWeaver Awesome! We had some discussion of adding a counter class to frequencyio. I'd probably do it as a separate module though. Maybe just `countio`?

I'd encourage you to make a PR and mark it as a draft. We're happy to see early code and help guide it along.

LearnWeaver, 2020-05-13:
Thanks for your help on this @tannewt

closed 2020-05-13 by LearnWeaver