adafruit/circuitpython #2818: Add transmit power control to BLE

Dar-Scott, 2020-04-27:
This would help in developing anonymous exposure notification apps to contain Covid-19 (privacy-preserving contact tracing apps) in testing and measurements–and perhaps even in devices.

In characterizing RSSI for different phones, control over BLE transmit power in a nRF52 CircuitPython board would be handy. There has also been talk of doing that to help with distance estimation. 

Perhaps this could be a property of _bleio.Adapter. (And perhaps eventually added to the appropriate class in adafruit_ble.)

The units should be physical if possible, such as dBm. If that can't be so, then ratios should work right. Simply being monotonic is not as useful. 

I believe the default is 0 dBM for nRF52840 soft device, but that is just a guess. I think that is fine for an initial value.

(I'm a Nordic noob and still sophomoric on BLE, so please take that into consideration.)

I'm on the [Covid Watch](covid-watch.org) team and I collaborate with other teams.

dhalbert, 2020-04-27:
The Nordic SoftDevice API allows separate power choices per connection and also per advertising setup. This flexibility is great, but is a complication, so we have to figure out where to pass in optional power levels.

Dar-Scott, 2020-04-27:
Yeah, I was wondering about separate power. I was a little timid and didn't want to ask for too much. The separate power might be great for emulation.

kevinjwalters, 2020-05-10:
A decision would be needed on how to handle/precisely define the existing partially implemented property of `BLERadio.tx_power` in https://github.com/adafruit/Adafruit_CircuitPython_BLE/blob/b08096f5eadc3bcf27461340e97d66d791073c05/adafruit_ble/__init__.py#L305-L312

dhalbert, 2020-05-10:
Yes, `BLERadio.tx_power` is too broad as it stands. Maybe it could be the default power. We could add `.default_tx_power` as an alias for now, deprecate `tx_power` in the doc, and then remove it in 6.0.0. Or, since it's not really implemented, probably nobody uses it, and we could just drop it or cause it to throw an error now on "get", since it already throws an error on "set".

Dar-Scott, 2020-05-13:
I'm using _bleio directly and not using BLERadio. Perhaps I should learn how to use the latter. However, at this time I am not using BLERadio.tx_power.

dhalbert, 2020-05-13:
@Dar-Scott `BLERadio.tx_power` doesn't do anything right now anyway :). It returns 0dBm and you can't set it.

tannewt, 2021-07-13:
#3493 allows for changing the tx power of an advertisement. It does not carry over to any connections (we reset to 0 dbm for connections) but this should satisfy this issue.

closed 2021-07-13 by tannewt