adafruit/circuitpython #2826: Add channel index to received advertisements in BLE

Dar-Scott, 2020-04-29:
One approach considered by the Covid Watch distance estimation team is separate consideration of RSSI based on the channel the advertisement came in on. The Covid Watch anonymous exposure notification app needs to make good exposure estimations. CircuitPython is used with small boards for experiments and testing.

The change would be in _bleio.

Perhaps _bleio.ScanEntry could have a new property called channel. It might be obtained from this (as suggested by another): 

https://infocenter.nordicsemi.com/topic/com.nordic.infocenter.s140.api.v6.1.1/structble__gap__evt__adv__report__t.html#a7620e6f82c248bc997b212037f6ce472

kevinjwalters, 2020-05-14:
Not directly related to this but is there any way to control or influence what channel an `Advertisement` goes out on?

dhalbert, 2020-05-14:
> Not directly related to this but is there any way to control or influence what channel an `Advertisement` goes out on when it's _transmitted_?

It is [possible](https://infocenter.nordicsemi.com/index.jsp?topic=%2Fcom.nordic.infocenter.s140.api.v6.1.1%2Fgroup___b_l_e___g_a_p___f_u_n_c_t_i_o_n_s.html&anchor=ga9969047f4e7485c3f856c841978cc31a) to restrict advertising to only some of the main primary advertising channels 37-39. The secondary channels cannot be turned off.

kevinjwalters, 2020-05-14:
I've seen rssi stepping which suggests there's a 5 second timer on changing channel for sending Advertisements over 37-79.

It might be useful to drop that momentarily that from 5s to 1s for some code I'm writing, I'm still investigating this so it may not help!

Dar-Scott, 2020-05-14:
Maybe the scan interval can beat against the advertisement interval in a way that you get the pattern you want. That seems hard. It might be easiest to filter the received packets. If you are looking at RSSI and distance for individual channels, I can point out some papers.

kevinjwalters, 2020-05-20:
I've failed to get my friend's children interested in RSSI measurements, unfortunately. My personal suspicious is PRNG would benchmark well against any measurement technique.

I've just taken a picture of what I see from an Android tablet showing a CPB advertising with the classic three 5 seconds steps: [RSSI variation with Bluetooth LE Advertisement](https://forums.adafruit.com/viewtopic.php?f=53&t=165613).

Dar-Scott, 2020-07-05:
I have been able to get a crude channel estimate by measuring the delta monotonic_ns() from the start of the scan in my GAEN code for supporting Covid Watch.