adafruit/circuitpython #2830: Default stack size on stm, i.mx much smaller than nRF, samd51

jepler, 2020-04-29:
Due to hitting the stack overflow error, I noticed that the stack size is much smaller (2kB) than nRF, samd51 (24kB).

My particular code is work in progress RGBMatrix, so the stack-overflowing code is not something that you can easily reproduce.

I tried increasing it to 0x6000 (24kB) by boot.py: 
```import supervisor
supervisor.set_next_stack_limit(0x6000)
```
but this leads to crashing errors!  0x2000 works, and I don't get the errors anymore.   However
 * default stack size should probably be more like other systems
 * why does stack limit 0x6000 crash?!

hierophect, 2020-04-29:
I'm concerned this may be the result of something being carried over from Atmel-SAMD in the very early stages of the port. I'll check it out.

hierophect, 2020-05-15:
This is resolved on STM32, investigating for i.MX

arturo182, 2020-05-18:
Which PR was this fixed in stm, what was the root cause?

hierophect, 2020-05-19:
@arturo182 temporary fix was in #2866, cause was _minimum_stack_size being set very small, and the "default" stack size is set based on minimum size. I haven't dug much deeper on the issue and it needs a proper, informed fix before I would consider closing this issue. I'll check it out for the i.MX as soon as I'm done with pin claiming (should be today).

ladyada, 2020-06-16:
@hierophect hihi can we bump the STM stack size or was that done in a 5.4.x release? a customer is hitting the stack limit

hierophect, 2020-06-16:
@ladyada yah STM32 stack size was bumped in 5.4.x to 24kb. I'd like to revisit it though and figure out why it's not dynamic.

hierophect, 2020-06-16:
Fix for stm32 was in #2866. Issue isn't closed since it includes the i.MX, which I haven't fully investigated yet (I know it isn't the same trivial fix that STM32 needed).

ladyada, 2020-06-16:
@hierophect ok so they just need to use latest 5.4? ill let em know!

hierophect, 2020-06-16:
@ladyada yeah let me know if it works! If anything goes wrong I'll put it at the top of my todo.

tannewt, 2020-06-16:
We don't need a dynamic stack when it's in a different memory region (DTCM or CCM) than the heap (DMA-able RAM).

hierophect, 2020-06-16:
@tannewt yeah I'd probably just add a check to exclude the F7 and H7 chips

tannewt, 2020-07-14:
@hierophect This done?

hierophect, 2020-07-14:
@tannewt still need to figure out how to fix it for i.MX, I can spend some time on that this week if you like.

tannewt, 2020-07-14:
No, it can wait. Please get to ESP. I'll change the issue title.

PR #7767 (2023-03-22): Make set_stack_limit respect fixed stack
Fixes #2830
commit 0eb08509 (2023-03-22): Make set_stack_limit respect fixed stack

Fixes #2830